JuliaRobotics/IncrementalInference.jl #904: Check examples and update if needed

Affie, 2020-09-17:
- [ ] ApproximateConvolution
 - [ ] BayesTreeIllustration
 - [ ] FixedPointIllustrationsSquare
 - [ ] IllustrateAutoInit
 - [ ] MultiHypo2Door
 - [ ] MultiHypo3Door
 - [ ] RobotFourDoor
 - [ ] SquareRootTypes
 - [ ] TexTreeIllustration
 - [ ] TreeAnalysis
 - [ ] squarefixedpoint

dehann, 2020-09-18:
Great, thanks for doing this!  FYI, we should add Square Root as intro example on CaesarDocs too.

Affie, 2021-02-09:
We need a better solution to keep the examples up to date. 
Ideas:
1. We can test it as a separate CI action to at least know when it's broken.
2. We can use documented and test it as examples. 
3. Maybe we should reduce it to a few (more manageable) key concepts. 

I would say 1 and possibly 3.

dehann, 2021-02-10:
I would say definitely all three, maybe better to work along priorities.  Let me try do something about 3 here quick.  I have never done 2, but know it can be done.  And 1. is good option too, and go there if easier.