JuliaRobotics/IncrementalInference.jl #910: Consolidate btnd.solvableDims into LikelihoodMessage

dehann, 2020-09-19:
https://github.com/JuliaRobotics/IncrementalInference.jl/blob/c7b3c3d91ae586c63d68f51b15e839ef7422cb7b/src/JunctionTreeTypes.jl#L347

dehann, 2020-10-02:
Okay, so logging how to solve this one.  Each sibling has a possible solvableDim per variable in the clique.  

The rule of thumb here is that if the solve were to occur one sibling at a time, then the last sibling would have the most information available to resolve its estimates;  simultaneously, the first sibling (and intermediate ones) may or may not have the required information from priors yet propagated to that part of the factor graph (from any prior and through measurement functions).

Given the current level of solved-ness of each variable (during init) some cliques are further away from being solved than others.  This information needs to be communicated to each sibling somehow, so that the best order of precedence in various situations can be resolved.  This communication should happen according to tree construction and messages passed along the edges only.

dehann, 2020-10-03:
The old update call is:
```julia
sdims = getCliqVariableMoreInitDims(csmc.cliqSubFg, csmc.cliq)
updateCliqSolvableDims!(csmc.cliq, sdims, csmc.logger)
```

Affie, 2020-10-06:
I'm still not completely following with `solvableDims`
Say we work with hex but with priors on x3 and x5 on this tree:
[image]

5 and 6 can both do an upinit, why can't they go at the same time?
They have x4 in common so is the information not best mixed in the parent (2)?

dehann, 2020-10-07:
still wip to get the new code to act the same.

dehann, 2020-10-07:
Hi @Affie , check out the previous discussion in #344

dehann, 2020-10-23:
not going to make this 

> at the same

and parking sibling cascade order in #954 (old fetch model).

PR #941 (2020-10-02): intermediate code refactor updates, towards #910
part of but does not close #910
commit 65a77f1b (2020-10-07): Merge pull request #953 from JuliaRobotics/fix/4Q20/910

WIP to fix #910
closed 2020-10-23 by dehann