JuliaRobotics/IncrementalInference.jl #924: parent starting down solve too early bug

dehann, 2020-09-27:
When solving `Caesar1D` or `Hex` with tree init, clique 1 starts downsolve before clique 2 completes it's upsolve

dehann, 2020-10-08:
Hi @Affie , I was wondering if you could have a go at this issue perhaps please?  Should be a basic redirect in old fetch CSM when doing hex init.  You can just work from the current `testBasicTreeInit.jl` test file?  

Aside: I think most of #954 will only happen in v0.17 timeframe.  That is a larger job than I previously thought -- but at least a bit more standard CSM structure as part of the #754/#910 effort.

Affie, 2020-10-08:
> Should be a basic redirect in old fetch CSM when doing hex init

I didn't quite understand this comment but went with something similar to the take! CSM. 
see #958

PR #958 (2020-10-08): Only downsolve when all children finished upsolve
close #924

Tests:
- Passed RoME tests locally
- Passed with #957 also locally
commit 1d0abc2a (2020-10-08): Only downsolve when all children finished upsolve 
close #924
closed 2020-10-08 by dehann