JuliaRobotics/IncrementalInference.jl #935: DOCS on Caesar.jl on how to do a parametric solve

dehann, 2020-10-01:
Need some docs to show how current parametric solve can be called.  Will do after CSM consolidation is complete (#855 ).  Will help address questions like JuliaRobotics/Caesar.jl#140

dehann, 2020-11-20:
still experimental.

Affie, 2021-02-01:
Capturing documentation part of RoME382 here: https://github.com/JuliaRobotics/RoME.jl/issues/382#issue-791284987

should write up how `getParametricMeasurement` works. The fallback is to use the .Z or .z field, with a warning (warn once) and link to the documentation (TODO).

dehann, 2021-02-01:
Sounds good thanks!

Affie, 2021-02-01:
Example of `getParametricMeasurement` here:
https://github.com/JuliaRobotics/RoME.jl/blob/1b62e23d81dcd4cea427a3f31820708266ab8480/src/factors/BearingRange2D.jl#L25-L32

dehann, 2021-03-27:
Note API change in v0.22.0 to `solveGraphParametric!` instead

closed 2021-03-27 by dehann