JuliaRobotics/IncrementalInference.jl #950: Manual init cliqSubFg in addLikelihoodsDifferential!

Affie, 2020-10-07:
Manual initialization in `addLikelihoodsDifferential!` overwrites the clique's own initialization and can cause the clique to take longer to converge. 
A possible solution is to use a separate empty fg.

https://github.com/JuliaRobotics/IncrementalInference.jl/blob/34eb45287466fce16cb3f89bbad7327f2bef8789/src/TreeMessageUtils.jl#L154-L165

Affie, 2020-10-07:
A simple test with 2 cliques and forcing the wrong init in the child. (NOTE: it will converge to the correct values after a few solves) 
The expected answer is the label's numeric part, ie. :x1 should be at 1.0

Results before:
```julia
┌ Info: lm0: 
└   ppe = -0.48084621051950643
┌ Info: x0: 
└   ppe = -0.29508540575443576
┌ Info: x1: 
└   ppe = 0.5962592233887228
┌ Info: x2: 
└   ppe = 1.5617661757826118
```
Results after adding and using `cliqSubFG` with a temporary `tfg`
```julia
function addLikelihoodsDifferential!(msgs::LikelihoodMessage, cliqSubFG::AbstractDFG, tfg::AbstractDFG=initfg() )
```

```julia
┌ Info: lm0: 
└   ppe = -0.016852379645236535
┌ Info: x0: 
└   ppe = 0.007672181464320319
┌ Info: x1: 
└   ppe = 1.004402507245821
┌ Info: x2: 
└   ppe = 1.9836818739182118
```

dehann, 2020-10-07:
Trying to parse .. do you mean that deconvolutions used to construct LIKELIHOOD_DIFFERENTIAL should ignore the the current variable estimates and always happen at a predetermined location?  Not sure how you are using `tfg` in the general solve case?

Affie, 2020-10-07:
> Trying to parse .. do you mean that deconvolutions used to construct LIKELIHOOD_DIFFERENTIAL should ignore the current variable estimates and always happen at a predetermined location on the domain?

No, when it is calculated, the variable in the subgraph is manually initialized to the value of the message. ln 164 above `initManual!(tfg, label, manikde!(val)) `
That should be independent of the clique subgraph as not to overwrite any existing value.

> Not sure how you are using tfg in the general solve case?

What is the general solve case? In the above function, `tfg` should be a temporary empty fg to add the message variables to, used to calculate the deconvolution. 

From my tests, it seems to work much better that way with bad init values in messages. 

I will send pr soon. Was just a bit busy.

PR #952 (2020-10-07): Use empty FG in addLikelihoodsDifferential!
close  #950
closed 2020-10-07 by dehann