JuliaRobotics/IncrementalInference.jl #982: Issue solving Hexagonal

dehann, 2020-10-20:
@jim-hill-r found on Hexagonal:

```julia
[ Info: Ensure variables are all initialized (graphinit)
[ Info: Solving over the Bayes (Junction) tree.
Building Bayes net from cloud...
Find potential functions for each clique
[ Info: Bayes Tree Complete
[ Info: Do tree based init-inference on tree
BoundsError: attempt to access 2-element Array{Symbol,1} at index [Bool[]]
Stacktrace:
 [1] throw_boundserror(::Array{Symbol,1}, ::Tuple{Base.LogicalIndex{Int64,BitArray{1}}}) at ./abstractarray.jl:541
 [2] checkbounds at ./abstractarray.jl:506 [inlined]
 [3] _getindex at ./multidimensional.jl:742 [inlined]
 [4] getindex(::Array{Symbol,1}, ::BitArray{1}) at ./abstractarray.jl:1060
 [5] getCliqVarIdsPriors(::IncrementalInference.TreeClique, ::Array{Symbol,1}, ::Bool) at /root/.julia/packages/IncrementalInference/XNguP/src/JunctionTree.jl:1164
 [6] getCliqVarInitOrderUp(::BayesTree, ::IncrementalInference.TreeClique) at /root/.julia/packages/IncrementalInference/XNguP/src/TreeBasedInitialization.jl:33
 [7] tryUpInitCliq_StateMachine(::CliqStateMachineContainer{BayesTreeNodeData,CloudGraphsDFG{SolverParams},LightDFG{SolverParams,DFGVariable,DFGFactor},BayesTree}) at /root/.julia/packages/IncrementalInference/XNguP/src/CliqueStateMachine_fetch.jl:456
 [8] (::StateMachine{CliqStateMachineContainer})(::CliqStateMachineContainer{BayesTreeNodeData,CloudGraphsDFG{SolverParams},LightDFG{SolverParams,DFGVariable,DFGFactor},BayesTree}, ::Nothing; pollinterval::Float64, breakafter::Function, verbose::Bool, verbosefid::Base.PipeEndpoint, verboseXtra::Symbol, iterlimit::Int64, injectDelayBefore::Nothing, recordhistory::Bool, housekeeping_cb::IncrementalInference.var"#506#508"{IncrementalInference.TreeClique}) at /root/.julia/packages/FunctionalStateMachine/2JZFG/src/StateMachine.jl:82
 [9] cliqInitSolveUpByStateMachine!(::CloudGraphsDFG{SolverParams}, ::BayesTree, ::IncrementalInference.TreeClique, ::Nothing; N::Int64, verbose::Bool, verbosefid::Base.PipeEndpoint, oldcliqdata::BayesTreeNodeData, drawtree::Bool, show::Bool, incremental::Bool, limititers::Int64, upsolve::Bool, downsolve::Bool, recordhistory::Bool, delay::Bool, injectDelayBefore::Nothing, logger::Base.CoreLogging.SimpleLogger) at /root/.julia/packages/IncrementalInference/XNguP/src/SolverAPI.jl:64
 [10] tryCliqStateMachineSolve!(::CloudGraphsDFG{SolverParams}, ::BayesTree, ::Int64, ::Nothing; verbose::Bool, verbosefid::Base.PipeEndpoint, N::Int64, oldtree::BayesTree, drawtree::Bool, limititers::Int64, downsolve::Bool, incremental::Bool, injectDelayBefore::Nothing, delaycliqs::Array{Symbol,1}, recordcliqs::Array{Symbol,1}) at /root/.julia/packages/IncrementalInference/XNguP/src/SolverAPI.jl:140
 [11] (::IncrementalInference.var"#522#525"{BayesTree,Bool,Base.PipeEndpoint,Bool,Int64,Bool,Bool,Array{Symbol,1},Array{Symbol,1},CloudGraphsDFG{SolverParams},BayesTree,Nothing,Int64,Int64,Nothing})() at ./task.jl:356┌ Warning: printCliqHistorySummary -- No CSM history found.
└ @ IncrementalInference ~/.julia/packages/IncrementalInference/XNguP/src/TreeDebugTools.jl:206
```

Deps:
- `Pkg.add(PackageSpec(name="DistributedFactorGraphs", version="0.10.6"));`
- `Pkg.add(PackageSpec(name="IncrementalInference", version="0.16"));`
- `Pkg.add(PackageSpec(name="RoME", version="0.9.2"));`

dehann, 2020-10-20:
this might be resolved with #979 -- will do backport for IIF v0.16.2

Affie, 2020-10-20:
I didn't get this error on master. Therefore, a new release should fix this

dehann, 2020-10-21:
thanks again!

closed 2020-10-21 by dehann