JuliaRobotics/IncrementalInference.jl #990: [Feature] Maybe only use clique in CSMC, not full tree

Affie, 2020-10-22:
Ok, so we can reevaluate in the future:
If we remove the tree from CSMC and only keep a clique, then we can copy the channels from the tree to the CSMC instead of the full parent and child cliques. 
It works by reference so the only difference will be with debugging as deep-copies are made.

_Originally posted by @Affie in https://github.com/JuliaRobotics/IncrementalInference.jl/issues/675#issuecomment-714306518_

dehann, 2020-10-22:
I think it would be good to ditch csmc.tree if able, as long as we keep the ability to do #742

dehann, 2021-01-02:
One reason for keeping `csmc.tree` is for debugging, to draw the entire tree state at each step in multithreaded system.  So let's keep tree for now, but what we can do is make sure that we are only using `csmc.clique`, `csmc.parent`, `csmc.children` fields during calculations.  We can avoid pointing back the the `tree` during computations.

This is not a priority now, but later perhaps we can improve the BayesTree api to have only a "shallow" version of the tree for visualization / debugging purposes, much like `DFGVariableSummary`.