JuliaRobotics/IncrementalInference.jl #998: Bug - Mixture Prior is not an AbstractPrior

Affie, 2020-10-24:
```julia
p = Mixture(Prior, [Normal(1,1),Normal(1,1)], [0.5, 0.5] )
isPrior == fasle
p isa AbstractPrior == false
```

This makes functions such as `lsfPriors(subfg)` not find it.

Affie, 2020-10-25:
One possible solution is:
DFG
```julia
function isPrior(dfg::AbstractDFG, fc::Symbol)
  fco = getFactor(dfg, fc)
  isPrior(getFactorType(fco))
end

function isPrior(::AbstractPrior)
  return true
end

function isPrior(::AbstractRelative)
  return false
end
```

IIF
```julia
function DistributedFactorGraphs.isPrior(::Mixture{N, F, S, T}) where {N,F,S,T}
  return F <: AbstractPrior
end
```

Affie, 2020-10-25:
@dehann let me know if this fits in with Mixtures, then I'll update DFG and IIF.

dehann, 2020-10-26:
thanks looks okay.  Just reaffirming from #1000.  Mixture can be any FunctorInferenceType, ie AbstractPrior or either of the AbstractRelaives.  this looks specific to issue is on AbstractPrior and suggested does not seem to impede AbstractRelative so looks good to me, thanks!

PR #1000 (2020-10-26): Mixture isPrior
needs DFG #679 to be merged and will be tagged as v0.10.7
close #998
closed 2020-10-26 by Affie