JuliaRobotics/IncrementalInference.jl #1015: prepBatchTree! not drawing bayes net

lemauee, 2020-11-04:
Hi,

Using the drawbayesnet=true keyword option of prepBatchTree!() produces an error:
```julia
ERROR: LoadError: type LightDFG has no field bn
Stacktrace:
 [1] getproperty(::LightDFG{SolverParams,DFGVariable,DFGFactor}, ::Symbol) at ./Base.jl:33
 [2] buildTreeFromOrdering!(::LightDFG{SolverParams,DFGVariable,DFGFactor}, ::Array{Symbol,1}; drawbayesnet::Bool, maxparallel::Nothing, solvable::Int64) at /home/leopold/.julia/packages/IncrementalInference/JEqfk/src/JunctionTree.jl:613
 [3] prepBatchTree!(::LightDFG{SolverParams,DFGVariable,DFGFactor}; variableOrder::Nothing, variableConstraints::Array{Symbol,1}, ordering::Symbol, drawpdf::Bool, show::Bool, filepath::String, viewerapp::String, imgs::Bool, drawbayesnet::Bool, maxparallel::Nothing) at /home/leopold/.julia/packages/IncrementalInference/JEqfk/src/JunctionTree.jl:720
 [4] top-level scope at /home/leopold/Nextcloud/Studium/Masterarbeit/svn/julia/mmiSAM/examples/exampleFromPaperLargeIntrospection.jl:71
 [5] include_string(::Function, ::Module, ::String, ::String) at ./loading.jl:1088
in expression starting at /home/leopold/Nextcloud/Studium/Masterarbeit/svn/julia/mmiSAM/examples/exampleFromPaperLargeIntrospection.jl:71
```

I suspect something changed in the DFG Implementation that breaks this (maybe seldomly used ;)) option?

Best,
Leo

Affie, 2020-11-04:
Hi, thanks. This looks like it predates DFG.jl. I'll look into restoring this functionality.

dehann, 2020-11-12:
Hi, the best function to use is `resetBuildTree!`, also see documentation page here:
https://juliarobotics.org/Caesar.jl/latest/principles/bayestreePrinciples/

Not sure the bayes net drawing will work right now unfortunately.  We should probably suppress that option for the time being.  The example in the Documentation should be good enough to at least describe the process.  Similarly there are at least 3 or 4 literature references that describe the Bayes net and tree process quite well
https://juliarobotics.org/Caesar.jl/latest/refs/literature/

- https://juliarobotics.org/Caesar.jl/latest/principles/bayestreePrinciples/
- Kaess WAFR 2010
- WAFR 2020
- Fourie Thesis, Chapter 5

lemauee, 2020-11-12:
Thanks again for the hints. I recently did the fg->bn->bt process for the example from the 2016 paper using pen&paper and just wanted to check if my intermediate bayes net results were the same. I fully understand that you cannot keep every visualization alive if the backend changes :)

dehann, 2020-11-13:
all good!

closed 2020-11-13 by dehann