JuliaRobotics/IncrementalInference.jl #1035: some graphs cannot be initialized, need warning

dehann, 2020-12-08:
CSM is run-away if no initialization is possible in some factor graph fragments. eg

```julia
fg = initfg()

addVariable!(fg, :x0, ContinuousScalar)
addVariable!(fg, :x1, ContinuousScalar)
addVariable!(fg, :x2, ContinuousScalar)

addFactor!(fg, [:x0;:x1], LinearRelative(Normal()))
addFactor!(fg, [:x1;:x2], LinearRelative(Normal()))

solveTree!(fg)
```

which results in this:
```julia

julia> solveTree!(fg)
┌ Warning: #TODO Verify useMsgLikelihoods=false works correctly
└ @ IncrementalInference ~/.julia/dev/IncrementalInference/src/SolverAPI.jl:279
[ Info: Ensure variables are all initialized (graphinit)
[ Info: x0 is not initialized, and will do so now...
[ Info: try doautoinit! of x0
[ Info: init with useinitfct Symbol[]
[ Info: x1 is not initialized, and will do so now...
[ Info: try doautoinit! of x1
[ Info: init with useinitfct Symbol[]
[ Info: x2 is not initialized, and will do so now...
[ Info: try doautoinit! of x2
[ Info: init with useinitfct Symbol[]
[ Info: x0 is not initialized, and will do so now...
[ Info: try doautoinit! of x0
[ Info: init with useinitfct Symbol[]
[ Info: x1 is not initialized, and will do so now...
[ Info: try doautoinit! of x1
[ Info: init with useinitfct Symbol[]
[ Info: x2 is not initialized, and will do so now...
[ Info: try doautoinit! of x2
[ Info: init with useinitfct Symbol[]
[ Info: x0 is not initialized, and will do so now...
[ Info: try doautoinit! of x0
[ Info: init with useinitfct Symbol[]
[ Info: x1 is not initialized, and will do so now...
[ Info: try doautoinit! of x1
[ Info: init with useinitfct Symbol[]
[ Info: x2 is not initialized, and will do so now...
[ Info: try doautoinit! of x2
[ Info: init with useinitfct Symbol[]
[ Info: x0 is not initialized, and will do so now...
[ Info: try doautoinit! of x0
[ Info: init with useinitfct Symbol[]
[ Info: x1 is not initialized, and will do so now...
[ Info: try doautoinit! of x1
[ Info: init with useinitfct Symbol[]
[ Info: x2 is not initialized, and will do so now...
[ Info: try doautoinit! of x2
[ Info: init with useinitfct Symbol[]
[ Info: x0 is not initialized, and will do so now...
[ Info: try doautoinit! of x0
[ Info: init with useinitfct Symbol[]
[ Info: x1 is not initialized, and will do so now...
[ Info: try doautoinit! of x1
[ Info: init with useinitfct Symbol[]
[ Info: x2 is not initialized, and will do so now...
[ Info: try doautoinit! of x2
[ Info: init with useinitfct Symbol[]
[ Info: x0 is not initialized, and will do so now...
[ Info: try doautoinit! of x0
[ Info: init with useinitfct Symbol[]
[... 149 lines cut ...]
[ Info: CSM-4d Empty message on clique 1 frontals
[ Info: CSM-4d Empty message on clique 2 frontals
[ Info: CSM-4d Empty message on clique 1 frontals
[ Info: CSM-4d Empty message on clique 2 frontals
[ Info: CSM-4d Empty message on clique 1 frontals
[ Info: CSM-4d Empty message on clique 2 frontals
[ Info: CSM-4d Empty message on clique 1 frontals
[ Info: CSM-4d Empty message on clique 2 frontals
[ Info: CSM-4d Empty message on clique 1 frontals
[ Info: CSM-4d Empty message on clique 2 frontals
[ Info: CSM-4d Empty message on clique 1 frontals
[ Info: CSM-4d Empty message on clique 2 frontals
Solve Progress: approx max 48, at iter 854 	 Time: 0:00:00[ Info: CSM-4d Empty message on clique 1 frontals
[ Info: CSM-4d Empty message on clique 2 frontals
[ Info: CSM-4d Empty message on clique 1 frontals
[ Info: CSM-4d Empty message on clique 2 frontals
[ Info: CSM-4d Empty message on clique 1 frontals
[ Info: CSM-4d Empty message on clique 2 frontals
[ Info: CSM-4d Empty message on clique 1 frontals
[ Info: CSM-4d Empty message on clique 2 frontals
[ Info: CSM-4d Empty message on clique 1 frontals
[ Info: CSM-4d Empty message on clique 2 frontals
[ Info: CSM-4d Empty message on clique 1 frontals
[ Info: CSM-4d Empty message on clique 2 frontals
```

dehann, 2020-12-08:
Hi @Affie, not sure if you have time available but perhaps this is something you could look at please?  This issue does not occur inside a single clique tree, but when there is more than one clique over the  uninitializable fragment of the graph, this happens.  Also remember that for two graph fragments, this might only occur with one fragment while the other might solve just fine.

Affie, 2020-12-08:
Thanks for catching this, I'll have a look. 
How was it handled in old CSM?
Will `maxiters` be sufficient as stopping criteria, or should it be smarter and look at initialization?

Affie, 2020-12-08:
@dehann, what about this:

- add `max_init_retry` to csmc
- if the limit is reached a warning is thrown
- then `ERROR` state is propagated to cleanly exit all connected cliques.

dehann, 2020-12-08:
limit tiers sounds fine.  Again i would rather have a warning rather than an error.  it should just pass over the graph and not change its values and exit with a warning.

closed 2020-12-09 by dehann