JuliaRobotics/IncrementalInference.jl #1046: animateStateMachineHistoryIntervalCompound not found from csmAnimate

Affie, 2020-12-10:
```julia
ERROR: UndefVarError: animateStateMachineHistoryIntervalCompound not defined
Stacktrace:
 [1] csmAnimate(::IncrementalInference.MetaBayesTree, ::Dict{Int64,Array{NamedTuple{(:timestamp, :id, :f, :csmc),Tuple{Dates.DateTime,Int64,Function,CliqStateMachineContainer}},1}}; frames::Int64, interval::Int64, dpi::Int64, rmfirst::Bool, folderpath::String, fsmColors::Dict{Symbol,String}, defaultColor::String) at /home/johan/.julia/dev/IncrementalInference/src/TreeDebugTools.jl:711
 [2] top-level scope at REPL[92]:1
```

dehann, 2020-12-13:
The `Compound` functions here are pretty important so definitely need to get this fixed with the new `MetaBayesTree` for sure!

dehann, 2020-12-28:
might be related to #1066

closed 2020-12-28 by dehann