JuliaRobotics/IncrementalInference.jl #1053: Missing deprecation for `fetchMsgUpThis`

dehann, 2020-12-16:
What is the replacement function from v0.16 for `fetchMsgUpThis`?

Hi @Affie, I think you removed this function perhaps?

https://github.com/JuliaRobotics/IncrementalInference.jl/blob/9ad3160c87ce84609ba2a2a3c70135f5b214c172/src/TreeMessageUtils.jl#L464

```julia
#TODO function is currently broken as getUpMsgs does not exist, possibly deprecated?
# DF, I did some digging and got this far
# v0.14 @deprecate getUpMsgs(x...) getMsgsUpThis(x...)
#       @deprecate getMsgsUpThis(x...) getMsgUpThis(x...)
#         getMsgUpThis(cdat::BayesTreeNodeData) = fetch(getMsgUpChannel(cdat))
# v0.16 @deprecate getMsgUpThis(x...;kw...) fetchMsgUpThis(x...;kw...)
# fetchMsgUpThis was never properly deprecated, ask @Affie
```

dehann, 2020-12-28:
I think the replacing function might be `getMessageBuffer(cliq).upTx`

closed 2020-12-30 by dehann

Affie, 2021-01-14:
Hi @dehann, As far as I remember the decision was to move the fetch csm to the attic without deprecation.  
`IIF.getMessageBuffer(cliq).upTx` doesn't sound correct (and its not in use as far as I remember). Is it not closer to `IIF.getMessageBuffer(cliq).upRx` or was it the message that was sent out?

PR #1079 (2020-12-30): add depr for fetchMsgUpThis
close #1053