JuliaRobotics/IncrementalInference.jl #1055: Debug tools are broken on new CSM

dehann, 2020-12-17:
Basic debugging task does not work any more:

```julia
smtasks = Task[]
tree, smt, hist = IIF.solveTree!(fg; smtasks=smtasks, recordcliqs=ls(fg));

hists = fetchCliqHistoryAll!(smtasks);
printCSMHistoryLogical(hists)

julia> length(hists[1])
4

julia> printCSMHistoryLogical(hists)
    | 1   x2             | 2   x0             | 3   x4             | 4   x5             
----+--------------------+--------------------+--------------------+--------------------
ERROR: KeyError: key 1 => 1 not found
Stacktrace:
 [1] getindex at ./dict.jl:467 [inlined]
 [2] printHistoryLane(::Base.TTY, ::Int64, ::Array{NamedTuple{(:timestamp, :id, :f, :csmc),Tuple{Dates.DateTime,Int64,Function,CliqStateMachineContainer}},1}, ::Dict{Pair{Int64,Int64},Int64}) at /home/dehann/.julia/dev/IncrementalInference/src/TreeDebugTools.jl:355
 [3] printCSMHistoryLogical(::Dict{Int64,Array{NamedTuple{(:timestamp, :id, :f, :csmc),Tuple{Dates.DateTime,Int64,Function,CliqStateMachineContainer}},1}}, ::Base.TTY; order::Array{Int64,1}, printLines::UnitRange{Int64}) at /home/dehann/.julia/dev/IncrementalInference/src/TreeDebugTools.jl:464
 [4] printCSMHistoryLogical(::Dict{Int64,Array{NamedTuple{(:timestamp, :id, :f, :csmc),Tuple{Dates.DateTime,Int64,Function,CliqStateMachineContainer}},1}}, ::Base.TTY) at /home/dehann/.julia/dev/IncrementalInference/src/TreeDebugTools.jl:397 (repeats 2 times)
 [5] top-level scope at REPL[46]:1
```

cc @Affie

dehann, 2020-12-17:
xref #443

Affie, 2020-12-18:
Sorry, these were broken by #1047

dehann, 2020-12-18:
it's okay, think it is fixed now.  Added to the tests too with #1056.  Just want to make you aware in case things show up again in the future.

closed 2020-12-18 by dehann