JuliaRobotics/IncrementalInference.jl #1086: parseusermultihypo assertion error

lemauee, 2020-12-30:
Hi, 

The assertion `@assert sum(multihypo2) % 1 ≈ 0 `  in `parseusermultihypo` throws an error if the user input is slightly below some integer:

```
ERROR: LoadError: AssertionError: sum(multihypo2) % 1 ≈ 0[ Info: monitorCSMs: all tasks done

Stacktrace:
 [1] parseusermultihypo(::Array{Float64,1}, ::Float64) at /home/leopold/.julia/packages/IncrementalInference/JEqfk/src/FactorGraph.jl:544
```

Which can be easily reproduced on the repl:

```julia-repl
julia> failMH = measurements["lm"]["jLm"]["p"][12]
2×1 Array{Float64,2}:
 0.4958648003278645
 0.5041351996721354

julia> res = sum(failMH) % 1
0.9999999999999999

julia> res ≈ 0
false
```

Modulo of some value slightly below some integer will always be 0.999... and not 0.

Perhaps a check function like 

```julia
function isapproxinteger(x)
	return x ≈ round(x)
end
```

used in an assertion like 

```julia
@assert isapproxinteger(sum(multihypo2))
```

would solve the problem, in the meantime i can always correct my probability input using 

```julia
p ./= sum(p)
```
but a more numerical-error-tolerant assertion would be good anyways.

P.S.: Using `≈` in against 0 is like using `==` (https://docs.julialang.org/en/v1/base/math/#Base.isapprox):
Note that x ≈ 0 (i.e., comparing to zero with the default tolerances) is equivalent to x == 0 since the default atol is 0. In such cases, you should either supply an appropriate atol (or use norm(x) ≤ atol) or rearrange your code (e.g. use x ≈ y rather than x - y ≈ 0). It is not possible to pick a nonzero atol automatically because it depends on the overall scaling (the "units") of your problem: for example, in x - y ≈ 0, atol=1e-9 is an absurdly small tolerance if x is the radius of the Earth in meters, but an absurdly large tolerance if x is the radius of a Hydrogen atom in meters.

dehann, 2020-12-30:
Hi @lemauee , thanks for reporting and suggesting.  Please see #1087 (v0.19.x) and backport #1088 (v0.18.2).

That should solve the issue you're seeing right?  I also added a better assert error message for when this happens again:
https://github.com/JuliaRobotics/IncrementalInference.jl/blob/03a48ea1de59687e12a29013e05c72529f560f12/src/FactorGraph.jl#L580

I also added a quick test here:

https://github.com/JuliaRobotics/IncrementalInference.jl/blob/03a48ea1de59687e12a29013e05c72529f560f12/test/testmultihypothesisapi.jl#L267

dehann, 2020-12-30:
expanded to test slightly above and below Int

PR #1088 (2020-12-30): fix multihypo tolerance, needs backport
resolve #1086, back port of #1087
commit 278aef62 (2020-12-30): fix multihypo tolerance, needs backport

resolve #1086
PR #1087 (2020-12-31): fix multihypo tolerance, needs backport
resolve #1086
commit 03a48ea1 (2020-12-31): fix multihypo tolerance, needs backport

resolve #1086
closed 2020-12-31 by dehann