JuliaRobotics/IncrementalInference.jl #1120: Does not yet run Julia 1.6-beta1, typename error

dehann, 2021-01-09:
Regular IIF and RoME tests produce similar errors with `typename(<SOMEFACTOR>) not defined`.  Here is a IIF test failure:
```julia
UndefVarError: typename(LinearRelative) not defined
Stacktrace:
 [1] _findSubgraphsFactorType(dfg_::LightDFG{SolverParams, DFGVariable, DFGFactor}, jointrelatives::Vector{NamedTuple{(:variables, :likelihood), Tuple{Vector{Symbol}, AbstractRelative}}}, separators::Vector{Symbol})
   @ IncrementalInference ~/.julia/packages/IncrementalInference/VkZO1/src/TreeMessageUtils.jl:184
 [2] _generateMsgJointRelativesPriors(cfg::LightDFG{SolverParams, DFGVariable, DFGFactor}, cliq::IncrementalInference.TreeClique)
   @ IncrementalInference ~/.julia/packages/IncrementalInference/VkZO1/src/TreeMessageUtils.jl:413
 [3] prepCliqueMsgUpConsolidated(subfg::LightDFG{SolverParams, DFGVariable, DFGFactor}, cliq::IncrementalInference.TreeClique, status::IncrementalInference.CliqStatus; logger::Base.CoreLogging.SimpleLogger, duplicate::Bool)
   @ IncrementalInference ~/.julia/packages/IncrementalInference/VkZO1/src/TreeMessageUtils.jl:679
 [4] postUpSolve_StateMachine(csmc::CliqStateMachineContainer{BayesTreeNodeData, LightDFG{SolverParams, DFGVariable, DFGFactor}, LightDFG{SolverParams, DFGVariable, DFGFactor}, MetaBayesTree})
   @ IncrementalInference ~/.julia/packages/IncrementalInference/VkZO1/src/CliqueStateMachine.jl:409
 [5] (::StateMachine{CliqStateMachineContainer})(userdata::CliqStateMachineContainer{BayesTreeNodeData, LightDFG{SolverParams, DFGVariable, DFGFactor}, LightDFG{SolverParams, DFGVariable, DFGFactor}, MetaBayesTree}, timeout::Int64; pollinterval::Float64, breakafter::Function, verbose::Bool, verbosefid::Base.TTY, verboseXtra::IncrementalInference.CliqStatus, iterlimit::Int64, injectDelayBefore::Nothing, recordhistory::Bool, housekeeping_cb::IncrementalInference.var"#378#380"{IncrementalInference.TreeClique})
   @ FunctionalStateMachine ~/.julia/packages/FunctionalStateMachine/2JZFG/src/StateMachine.jl:94
 [6] initStartCliqStateMachine!(dfg::LightDFG{SolverParams, DFGVariable, DFGFactor}, tree::MetaBayesTree, cliq::IncrementalInference.TreeClique, timeout::Int64; oldcliqdata::BayesTreeNodeData, verbose::Bool, verbosefid::Base.TTY, drawtree::Bool, show::Bool, incremental::Bool, limititers::Int64, upsolve::Bool, downsolve::Bool, recordhistory::Bool, delay::Bool, logger::Base.CoreLogging.SimpleLogger, solve_progressbar::Nothing, algorithm::Symbol)
   @ IncrementalInference ~/.julia/packages/IncrementalInference/VkZO1/src/CliqueStateMachine.jl:63
 [7] tryCliqStateMachineSolve!(dfg::LightDFG{SolverParams, DFGVariable, DFGFactor}, treel::MetaBayesTree, cliqKey::Int64, timeout::Int64; oldtree::MetaBayesTree, verbose::Bool, verbosefid::Base.TTY, drawtree::Bool, limititers::Int64, downsolve::Bool, incremental::Bool, delaycliqs::Vector{Symbol}, recordcliqs::Vector{Symbol}, solve_progressbar::Nothing, algorithm::Symbol)
   @ IncrementalInference ~/.julia/packages/IncrementalInference/VkZO1/src/SolverAPI.jl:110
 [8] (::IncrementalInference.var"#434#437"{MetaBayesTree, Bool, Bool, Base.TTY, Bool, Bool, Vector{Symbol}, Vector{Symbol}, Symbol, LightDFG{SolverParams, DFGVariable, DFGFactor}, MetaBayesTree, Int64, Nothing, Int64, Int64})()
   @ IncrementalInference ./task.jl:406
```

dehann, 2021-01-14:
The code breaking here is part of the new code towards resolving #1010

Affie, 2021-01-21:
Resolving this error breaks numerical tests.

PR #1122 (2021-01-21): _findSubgraphsFactorType use nameof
fix #1120 typename error, but tests still not passing
closed 2021-01-21 by Affie