JuliaRobotics/IncrementalInference.jl #1166: How best to reuse factors (CalcFactor)

Affie, 2021-02-15:
With reference to this issue: https://github.com/JuliaRobotics/IncrementalInference.jl/issues/467#issuecomment-769192310

Is this the best way (wrap it in another CalcFactor):
```julia
cfZij = CalcFactor( cfo.factor.Zij.mechanics, cfo.metadata, 0, length(cfo._legacyMeas), cfo._legacyMeas, cfo._legacyParams)
```
https://github.com/JuliaRobotics/RoME.jl/blob/f41342dd89c47c1a5e6699990889a7ff609f20e6/src/factors/flux/MixtureFluxPose2Pose2.jl#L150

A helper CalcFactor constructor might be handy if this is the preferred way.
We should also add it to documentation in Caeser

Affie, 2022-07-04:
What about this pattern:
```julia
function (cf::CalcFactor{<:Pose2Pose2})(X, p, q)
  return Pose2Pose2(X, p, q)
end

function (cf::CalcFactor{<:ScatterAlignPose2})(X, p, q)
   return Pose2Pose2(X, p, q)
end

@inline function (Pose2Pose2)(X, p, q)
  q̂ = allocate(q) 
  M = getManifold(Pose2)
  ϵ0 = getPointIdentity(M)
  exp!(M, q̂, ϵ0, X)
  Manifolds.compose!(M, q̂, p, q̂)   
  Xc = vee(M, q, log!(M, q̂, q, q̂))
  return Xc
end
```

dehann, 2022-07-05:
Oh that's cool, haven't seen that before!

Affie, 2022-07-05:
I don't know if it's a good idea though. I edited the code block to remove () I left from copy-paste. 
It's a little weird to have the same name as the constructor that isn't a constructor.

Affie, 2022-07-05:
Maybe just:
```
@inline function calcPose2Pose2(X, p, q) ...
```

dehann, 2022-07-05:
oh right, it's just the constructor handle.  Do we need this for the AD stuff?

Affie, 2022-07-06:
> oh right, it's just the constructor handle. Do we need this for the AD stuff?

 No we don’t. I was just thinking of reuse for  factors that we optimize.

dehann, 2022-07-07:
The reuse is very nice!  Okay, thanks just wanted to see if this helped avoid any of the calcfactor stuff when doing AD but sounds like the type-stability story is separate.