JuliaRobotics/IncrementalInference.jl #1170: solveCliq feature update

Affie, 2021-02-18:
Just documenting what we spoke about...
- record cliques replace old solveCliq function
- solve clique will do up/down
- in messages can be created from fg or input directly
- out messages will be output
- therefore, it becomes more of a debug feature on how a clique is solved with different messages
- we can add extra plumbing later to ease clique interaction.

Affie, 2021-02-24:
See up poc #1184

dehann, 2021-02-24:
Cool, also we should be able to step through the local CSM of that clique afterward.  So a mechanism to extract the CSM history vector.

Affie, 2021-02-25:
@dehann, what do we do with the current `solveCliq!`
1. deprecate
2. call `solveCliqUp!` then `solveCliqueDown!` and combine the results

my vote is for 1, but don't mind either.

dehann, 2021-02-25:
Hi @Affie ,

Lets just deprecate and keep moving.  Perhaps we should just put that in the NEWS.md file in case someone is using it.

Thanks

Affie, 2021-02-25:
Ok, I deprecated it cleanly with a warning, so news shouldn't be necessary.

PR #1186 (2021-02-25): add solveCliqDown!
close #1170
closed 2021-02-25 by Affie