JuliaRobotics/IncrementalInference.jl #1174: Restore API for `getMsgUp(cliq)`

dehann, 2021-02-20:
Hi @Affie ,

Could you perhaps re-add functions for getting the latest up (similar for down) message from a clique please.  The usage used to be:
```julia
fg = ...
tree, _, = solveTree!(fg);

upmsg = getMsgUp(getClique(tree, :1))
```

Looks like all those functions were removed during #459.  The idea would be that we can manually inspect and/or use with `solveCliq!` for debugging.

Thanks,
Dehann

Affie, 2021-02-21:
Will these work? I added, but it felt unnecessary so commented it out. 
I’ll just re add and and make sure DownRx is populated. 

https://github.com/JuliaRobotics/IncrementalInference.jl/blob/a2ef785854b2b29b4b22b745d9ce21e1ea2f7ea4/src/TreeMessageAccessors.jl#L120-L132

dehann, 2021-02-21:
Hi Johan, yes the commented out ones look like the right thing:
https://github.com/JuliaRobotics/IncrementalInference.jl/blob/a2ef785854b2b29b4b22b745d9ce21e1ea2f7ea4/src/TreeMessageAccessors.jl#L131

Also see RP 155 liked just above

dehann, 2021-02-21:
should also add them to the tests so that we can make sure the API for messages on the tree remains available.

PR #1180 (2021-02-24): Add tree message getters
- close #1174
- fix compare likelihoodMessage bug
closed 2021-02-24 by Affie