JuliaRobotics/IncrementalInference.jl #1196: drawGraph does not allow relative path

Affie, 2021-03-01:
> > Just as a note for writing the somehow protechted ARGS, this is the stuff I do now:
> > ```
> > julia> newARGS=["--path", "tmp2", "--file", "sigmaTrajTf-0-000-0-000-0-003_sigmaLmBearingAndRanging-0-002-0-001.mat", "--trajKeys", "tf", "--lmKeys", "bearingAndRanging", "--startPoseIdx", "1", "--endPoseIdx", "150", "--startPoseVal", "[7.15793412168699;3.38794983135837;-1.36840501375598]", "--plotSaveFinal", "1", "--plotSaveIter", "1", "--nRuns", "1", "--suffix", "variables", "--useMsgLikelihoods", "0", "--nullhypo", "0.000000", "--tukey", "15.000000", "--nKernels", "100", "--spreadNH", "3.000000", "--inflation", "3.000000"]
> > julia> empty!(ARGS)
> > julia> for arg in newARGS; push!(ARGS,arg); end
> > ```
> > 
> > 
> > A new session with drawtree/showtree/async off is running now, but I still have the old one online.
> > EDIT: when running the new session on the same machine with the old one still on another screen, I get this error:
> > ```julia
> > ERROR: LoadError: TaskFailedException:
> > IOError: mkdir: permission denied (EACCES)
> > Stacktrace:
> >  [1] uv_error at ./libuv.jl:97 [inlined]
> >  [2] mkdir(::String; mode::UInt16) at ./file.jl:177
> >  [3] mkpath(::String; mode::UInt16) at ./file.jl:227
> >  [4] mkpath(::String; mode::UInt16) at ./file.jl:225 (repeats 4 times)
> >  [5] mkpath at ./file.jl:222 [inlined]
> >  [6] drawGraph(::LightDFG{SolverParams,DFGVariable,DFGFactor}; viewerapp::String, filepath::String, engine::String, show::Bool) at /home/leopold/.julia/packages/IncrementalInference/nmcd8/src/AdditionalUtils.jl:56
> >  [7] _dbgCSMSaveSubFG(::CliqStateMachineContainer{BayesTreeNodeData,LightDFG{SolverParams,DFGVariable,DFGFactor},LightDFG{SolverParams,DFGVariable,DFGFactor},MetaBayesTree}, ::String) at /home/leopold/.julia/packages/IncrementalInference/nmcd8/src/CliqStateMachineUtils.jl:75
> >  [8] buildCliqSubgraph_StateMachine(::CliqStateMachineContainer{BayesTreeNodeData,LightDFG{SolverParams,DFGVariable,DFGFactor},LightDFG{SolverParams,DFGVariable,DFGFactor},MetaBayesTree}) at /home/leopold/.julia/packages/IncrementalInference/nmcd8/src/CliqueStateMachine.jl:122
> >  [9] (::StateMachine{CliqStateMachineContainer})(::CliqStateMachineContainer{BayesTreeNodeData,LightDFG{SolverParams,DFGVariable,DFGFactor},LightDFG{SolverParams,DFGVariable,DFGFactor},MetaBayesTree}, ::Int64; pollinterval::Float64, breakafter::Function, verbose::Bool, verbosefid::Base.TTY, verboseXtra::IncrementalInference.CliqStatus, iterlimit::Int64, injectDelayBefore::Nothing, recordhistory::Bool, housekeeping_cb::IncrementalInference.var"#382#384"{IncrementalInference.TreeClique}) at /home/leopold/.julia/packages/FunctionalStateMachine/2JZFG/src/StateMachine.jl:94
> >  [10] initStartCliqStateMachine!(::LightDFG{SolverParams,DFGVariable,DFGFactor}, ::MetaBayesTree, ::IncrementalInference.TreeClique, ::Int64; oldcliqdata::BayesTreeNodeData, verbose::Bool, verbosefid::Base.TTY, drawtree::Bool, show::Bool, incremental::Bool, limititers::Int64, upsolve::Bool, downsolve::Bool, recordhistory::Bool, delay::Bool, logger::Base.CoreLogging.SimpleLogger, solve_progressbar::ProgressMeter.ProgressUnknown, algorithm::Symbol) at /home/leopold/.julia/packages/IncrementalInference/nmcd8/src/CliqueStateMachine.jl:63
> >  [11] tryCliqStateMachineSolve!(::LightDFG{SolverParams,DFGVariable,DFGFactor}, ::MetaBayesTree, ::Int64, ::Int64; oldtree::MetaBayesTree, verbose::Bool, verbosefid::Base.TTY, drawtree::Bool, limititers::Int64, downsolve::Bool, incremental::Bool, delaycliqs::Array{Symbol,1}, recordcliqs::Array{Symbol,1}, solve_progressbar::ProgressMeter.ProgressUnknown, algorithm::Symbol) at /home/leopold/.julia/packages/IncrementalInference/nmcd8/src/SolverAPI.jl:110
> >  [12] (::IncrementalInference.var"#439#442"{MetaBayesTree,Bool,Bool,Base.TTY,Bool,Bool,Array{Symbol,1},Array{Symbol,1},Symbol,LightDFG{SolverParams,DFGVariable,DFGFactor},MetaBayesTree,Int64,ProgressMeter.ProgressUnknown,Int64,Int64})() at ./task.jl:356
> > Stacktrace:
> > [ Info: monitorCSMs: all tasks done
> >  [1] sync_end(::Channel{Any}) at ./task.jl:314
> >  [2] macro expansion at ./task.jl:333 [inlined]
> >  [3] taskSolveTree!(::LightDFG{SolverParams,DFGVariable,DFGFactor}, ::MetaBayesTree, ::Int64; oldtree::MetaBayesTree, drawtree::Bool, verbose::Bool, verbosefid::Base.TTY, limititers::Int64, limititercliqs::Array{Pair{Symbol,Int64},1}, downsolve::Bool, incremental::Bool, multithread::Bool, skipcliqids::Array{Symbol,1}, recordcliqs::Array{Symbol,1}, delaycliqs::Array{Symbol,1}, smtasks::Array{Task,1}, algorithm::Symbol) at /home/leopold/.julia/packages/IncrementalInference/nmcd8/src/SolverAPI.jl:49
> >  [4] solveTree!(::LightDFG{SolverParams,DFGVariable,DFGFactor}, ::MetaBayesTree; timeout::Int64, storeOld::Bool, verbose::Bool, verbosefid::Base.TTY, delaycliqs::Array{Symbol,1}, recordcliqs::Array{Symbol,1}, limititercliqs::Array{Pair{Symbol,Int64},1}, injectDelayBefore::Nothing, skipcliqids::Array{Symbol,1}, eliminationOrder::Nothing, variableOrder::Nothing, eliminationConstraints::Array{Symbol,1}, variableConstraints::Nothing, smtasks::Array{Task,1}, dotreedraw::Array{Int64,1}, runtaskmonitor::Bool, algorithm::Symbol, multithread::Bool) at /home/leopold/.julia/packages/IncrementalInference/nmcd8/src/SolverAPI.jl:371
> >  [5] macro expansion at /home/leopold/Masterarbeit/svn/julia/mmiSAM/evaluation/solve2DIncremental.jl:156 [inlined]
> >  [6] macro expansion at ./timing.jl:233 [inlined]
> >  [7] top-level scope at /home/leopold/Masterarbeit/svn/julia/mmiSAM/evaluation/solve2DIncremental.jl:149
> >  [8] include(::String) at ./client.jl:457
> >  [9] top-level scope at REPL[4]:1
> > in expression starting at /home/leopold/Masterarbeit/svn/julia/mmiSAM/evaluation/solve2DIncremental.jl:131
> > ```
> > 
> > 
> > Should be related to the `getSolverParams(fg).dbg=true` still set.
> 
> Also appears on another machine.

At least I got this solved, it originated from me using a relative path as logpath for the fg (which works fine if not debugging ;)). What lead me to the solution was the error that one can take much more from in julia 1.6 than in 1.5.3:

```julia-repl
IOError: mkdir("/tmp2"; mode=0o777): permission denied (EACCES)
Stacktrace:
  [1] uv_error
    @ ./libuv.jl:97 [inlined]
  [2] mkdir(path::String; mode::UInt16)
    @ Base.Filesystem ./file.jl:179
  [3] mkpath(path::String; mode::UInt16)
    @ Base.Filesystem ./file.jl:230
  [4] mkpath(path::String; mode::UInt16)
    @ Base.Filesystem ./file.jl:228
  [5] mkpath
    @ ./file.jl:225 [inlined]
  [6] drawGraph(fgl::LightDFG{SolverParams, DFGVariable, DFGFactor}; viewerapp::String, filepath::String, engine::String, show::Bool)
    @ IncrementalInference ~/.julia/packages/IncrementalInference/8DImq/src/AdditionalUtils.jl:56
  [7] _dbgCSMSaveSubFG(csmc::WARNING: both ManifoldsBase and ApproxManifoldProducts export "vee"; uses of it in module IncrementalInference must be qualified
WARNING: both ManifoldsBase and ApproxManifoldProducts export "vee!"; uses of it in module IncrementalInference must be qualified
CliqStateMachineContainer{BayesTreeNodeData, LightDFG{SolverParams, DFGVariable, DFGFactor}, LightDFG{SolverParams, DFGVariable, DFGFactor}, MetaBayesTree}, filename::String)
    @ IncrementalInference ~/.julia/packages/IncrementalInference/8DImq/src/CliqStateMachineUtils.jl:77
  [8] buildCliqSubgraph_StateMachine(csmc::CliqStateMachineContainer{BayesTreeNodeData, LightDFG{SolverParams, DFGVariable, DFGFactor}, LightDFG{SolverParams, DFGVariable, DFGFactor}, MetaBayesTree})
    @ IncrementalInference ~/.julia/packages/IncrementalInference/8DImq/src/CliqueStateMachine.jl:122
  [9] (::StateMachine{CliqStateMachineContainer})(userdata::CliqStateMachineContainer{BayesTreeNodeData, LightDFG{SolverParams, DFGVariable, DFGFactor}, LightDFG{SolverParams, DFGVariable, DFGFactor}, MetaBayesTree}, timeout::Nothing; pollinterval::Float64, breakafter::Function, verbose::Bool, verbosefid::Base.TTY, verboseXtra::IncrementalInference.CliqStatus, iterlimit::Int64, injectDelayBefore::Nothing, recordhistory::Bool, housekeeping_cb::IncrementalInference.var"#382#384"{IncrementalInference.TreeClique})
    @ FunctionalStateMachine ~/.julia/packages/FunctionalStateMachine/2JZFG/src/StateMachine.jl:82
 [10] initStartCliqStateMachine!(dfg::LightDFG{SolverParams, DFGVariable, DFGFactor}, tree::MetaBayesTree, cliq::IncrementalInference.TreeClique, timeout::Nothing; oldcliqdata::BayesTreeNodeData, verbose::Bool, verbosefid::Base.TTY, drawtree::Bool, show::Bool, increment [... cut ...]
    @ IncrementalInference ./task.jl:406┌ Warning: printCliqHistorySummary -- No CSM history found.
└ @ IncrementalInference ~/.julia/packages/IncrementalInference/8DImq/src/TreeDebugTools.jl:211
┌ Warning: printCliqHistorySummary -- No CSM history found.
└ @ IncrementalInference ~/.julia/packages/IncrementalInference/8DImq/src/TreeDebugTools.jl:211
```

Maybe we should open an issue for the debugging not working correctly with a relative logpaht? Or is this expected to fail?

_Originally posted by @lemauee in https://github.com/JuliaRobotics/RoME.jl/issues/418#issuecomment-787510942_

Affie, 2021-03-01:
Here:
https://github.com/JuliaRobotics/IncrementalInference.jl/blob/179f61f633773b733de9d4748dd5ead74a0abf5b/src/AdditionalUtils.jl#L56

dehann, 2021-03-08:
think this should now be fixed

closed 2021-03-08 by dehann