JuliaRobotics/IncrementalInference.jl #1204: Potential bug in assembleHypothesesElements!

Affie, 2021-03-12:
@dehann , It looks like a `[x, l1, l2]` `multihypo = [1,0.5,0.5]`  only uses 33% of `x` when solving for `l1`, or `l2`
I would expect 50%
eg:
```julia
N = 100
mh = Categorical([0, 0.5,0.5])
certainidx, allelements, activehypo, mhidx = IIF.assembleHypothesesElements!(mh, N, 2, 3)

([1], Any[[2, 5, 8, 12, 14, 16, 17, 20, 21, 22  …  65, 66, 71, 81, 82, 91, 92, 93, 94, 99], Int64[], [1, 7, 13, 15, 18, 24, 25, 28, 29, 45  …  73, 75, 76, 77, 79, 80, 84, 85, 90, 96], [3, 4, 6, 9, 10, 11, 19, 31, 32, 33  …  78, 83, 86, 87, 88, 89, 95, 97, 98, 100]], Any[(0, [2]), (1, [1, 2]), (2, [1, 2]), (3, [2, 3])], [2, 0, 3, 3, 0, 3, 2, 0, 3, 3  …  0, 0, 0, 0, 3, 2, 3, 3, 0, 3])

samples for 0: 30.599999999999998%
samples for 1: 0.0%
samples for 2: 32.7%
samples for 3: 36.7%

```

Sorry if I'm misunderstanding something but I thought I'd rather ask just in case.

Affie, 2021-03-12:
Also, should we perhaps use the nullhypo (spreadNH) case in initialization:
```julia
certainidx, allelements, activehypo, mhidx = IIF.assembleHypothesesElements!(mh, N, 1, 3, Bool[0, 0, 1])
#currently:
#samples from 0: 0.0%
#samples from 1: 0.0%
#samples from 2: 0.0%
#samples from 3: 100.0%

#suggested:
#samples from 0: 50.0%
#samples from 1: 0.0%
#samples from 2: 0.0%
#samples from 3: 50.0%
```

dehann, 2021-03-12:
Perhaps you missed in the the discussion.  We had forced a nullhypo case (0) previously, so currently you are effectively getting 60% nullhypo (0 + 2||3).  The bug is allocating the proper probability weight for "snap together case" in general n-ary use:

> samples for 0: 30.599999999999998%
samples for 1: 0.0%
samples for 2: 32.7%
samples for 3: 36.7%

Affie, 2021-03-13:
I didn’t miss it in the discussion. I understand that part. 
The issue is even if you give the probability of 0.999-0.001 it wil go to 33% nullhypo and 66% x