JuliaRobotics/IncrementalInference.jl #1205: Bug, `useMsgLikelihoods=true` does not propagate correct upward msgs

dehann, 2021-03-16:
When different factor types are used

MWE:
```julia
using RoME

fg = generateCanonicalFG_ZeroPose2()
getSolverParams(fg).useMsgLikelihoods = true

addVariable!(fg, :x1, Pose2)
addVariable!(fg, :l1, Point2)
addVariable!(fg, :l2, Point2)

addFactor!(fg, [:x0;:x1], Pose2Pose2(MvNormal(zeros(3),0.1*ones(3))))
addFactor!(fg, [:x1;:x2], Pose2Pose2(MvNormal(zeros(3),0.1*ones(3))))

addFactor!(fg, [:x0;:l1], Pose2Point2Range(Normal()) )
addFactor!(fg, [:x1;:l1], Pose2Point2Range(Normal()) )
addFactor!(fg, [:x2;:l2], Pose2Point2Range(Normal()) )

smtasks = Task[]

tree,_, = solveTree!(fg; smtasks, recordcliqs=ls(fg))
hists  = IIF.fetchCliqHistoryAll!(smtasks);

# get root clique sfg just before upsolve and after upward messages should have been added
sfg1_5 = hists[1][5].csmc.cliqSubFg

# should be the solveUp step in CSM
@assert :x1f1 in ls(sfg1_5) "must have upward marginal into root clique"
```

on IIF v0.21.4 one finds this upward message belief is missing in root clique (possibly others, haven't checked).

Issue found while debugging a test case from @doublestrong

cc'ing @Affie as well.

dehann, 2021-03-16:
I think this is a problem with CSM.  The up message has the `:x1` marginal prior (as it should):
```julia
julia> csmc1_5 = hists[1][5].csmc;

julia> IIF.getMessageUpRx(getClique(csmc1_5.tree,1))[2]
LikelihoodMessage:
status: IncrementalInference.UPSOLVED
belief: Dict{Symbol, TreeBelief}(:x1 => TreeBelief{Pose2}([0.5799465050123339 0.0031159899100484177 … 0.17795565811416986 -0.571620568726436; -0.08535986845527568 -0.5726356149322868 … 0.012895729380345392 0.10989757928228458; 0.4510448742936275 -0.2393183785017025 … 0.0642182613232917 0.09484169203741576], [0.10142532520298787; 0.14799256537210753; 0.12456705327172717], 9.0, Pose2(), (:Euclid, :Euclid, :Circular), 0.0))
variableOrder: Symbol[]
cliqueLikelihood: nothing
msgType: IncrementalInference.NonparametricMessage()
hasPriors: false
childSolvDims: Dict{Int64, Float64}()
jointmsg: 
 _MsgJointLikelihood:
  .relatives: 
  .priors: x1, 
```

But this prior does not show up in the clique subgraph:
```julia
drawGraph(csmc1_5.cliqSubFg, show=true);
```

[image]

dehann, 2021-03-16:
This if statement is likely the problem:
https://github.com/JuliaRobotics/IncrementalInference.jl/blob/22898c54084dbb251337f73d2d0f37cee8640556/src/TreeMessageUtils.jl#L440

Affie, 2021-03-18:
My guess would be that this issue causes the error in #1136

Affie, 2021-03-18:
I've looked at the example and by my logic it behaves as expected. 
There is no information in clique 2 and 3 that contributes the solution in the upward direction and only a down-solve is needed.

closed 2021-04-25 by dehann