JuliaRobotics/IncrementalInference.jl #1228: Why `getManifolds(vd::VariableNodeData)`

dehann, 2021-04-24:
Should `variableType` not just exist at DFGVariable level?  Why all the way down to VND.

dehann, 2021-04-24:
@Affie , do you perhaps remember why we made variableType a VND level thing?  My expectation was/is that each variable can only have one `variableType`.  Supersolve is for numerical differences only, not for mixing different `variableType`s.

dehann, 2021-04-25:
Going further, why did we have `VND{T <: InferenceVariable}`, and not just `DFGVariable{T}`?  This is more of a DFG consideration.

Affie, 2021-07-07:
It might just be historic since `softtype` used to be a field in vnd. With the replacement `variabletype` as a singleton type, there shouldn't be a reason to store the type itself. 
I can't see a reason it has to remain as a field. I don't know if the type is needed somewhrere though.