JuliaRobotics/IncrementalInference.jl #1232: tag v0.23.2

dehann, 2021-04-25:
@JuliaRegistrator register branch=release/v0.23

JuliaRegistrator, 2021-04-25:
Registration pull request created: [JuliaRegistries/General/35246](https://github.com/JuliaRegistries/General/pull/35246)

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if the [Julia TagBot GitHub Action](https://github.com/marketplace/actions/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v0.23.2 -m "<description of version>" f30d90ed4178d2b2b41804df5e565997badab683
git push origin v0.23.2
```

closed 2021-04-25 by dehann

dehann, 2021-04-25:
HI @jim-hill-r , release is done -- you can try `solveTree!(fg, solveKey=:curiousgeorge)`.  Note that does still require `getSolverParams(fg).graphinit=true`.  

Likely more work required to make this work under `treeinit`, but not necessary yet.  It will be easier to do that work once the following issues are done first:
- JuliaRobotics/DistributedFactorGraphs.jl#753,
- #1229