JuliaRobotics/IncrementalInference.jl #1239: Dimension issue in mixture

Affie, 2021-05-04:
Just logging a possible bug, I haven't confirmed if it is indeed a bug and what causes it.
This works:
```julia
fg = initfg()
addVariable!(fg, :x0, ContinuousEuclid(2))
f = addFactor!(fg, [:x0], Prior(MvNormal([0.8, 0.1], [0.4, 0.4])))
```

This errors
```julia
fg = initfg()
addVariable!(fg, :x0, ContinuousEuclid(2))
p = Mixture(Prior(I), [MvNormal([0.8, 0.1], [0.4, 0.4]), MvNormal([1.0, 0.1], [0.1, 0.1])], Categorical([0.1; 0.9]))
f = addFactor!(fg, [:x0], p)
```

stacktrace
```julia
ERROR: DimensionMismatch("tried to assign 2×1 array to 1×1 destination")
Stacktrace:
  [1] throw_setindex_mismatch(X::Matrix{Float64}, I::Tuple{Int64, Int64})
    @ Base ./indices.jl:193
  [2] setindex_shape_check
    @ ./indices.jl:253 [inlined]
  [3] _unsafe_setindex!(::IndexLinear, ::Matrix{Float64}, ::Matrix{Float64}, ::Base.Slice{Base.OneTo{Int64}}, ::Int64)
    @ Base ./multidimensional.jl:896
  [4] _setindex!
    @ ./multidimensional.jl:887 [inlined]
  [5] setindex!
    @ ./abstractarray.jl:1267 [inlined]
  [6] getSample(cf::CalcFactor{Mixture{2, Prior{Normal{Float64}}, (:c1, :c2), Tuple{DiagNormal, DiagNormal}}, FactorMetadata{Vector{DFGVariable{ContinuousEuclid{2}}}, Vector{Symbol}, Vector{Matrix{Float64}}, Nothing}, Tuple{Matrix{Float64}}, Vector{Matrix{Float64}}}, N::Int64)
    @ IncrementalInference ~/.julia/dev/IncrementalInference/src/Factors/Mixture.jl:85
  [7] sampleFactor
    @ ~/.julia/dev/IncrementalInference/src/CalcFactor.jl:55 [inlined]
  [8] calcZDim
    @ ~/.julia/dev/IncrementalInference/src/FactorGraph.jl:623 [inlined]
  [9] prepgenericconvolution(Xi::Vector{DFGVariable{ContinuousEuclid{2}}}, usrfnc::Mixture{2, Prior{Normal{Float64}}, (:c1, :c2), Tuple{DiagNormal, DiagNormal}}; multihypo::Nothing, nullhypo::Float64, threadmodel::Type, inflation::Float64)
    @ IncrementalInference ~/.julia/dev/IncrementalInference/src/FactorGraph.jl:648
 [10] getDefaultFactorData(dfg::LightDFG{SolverParams, DFGVariable, DFGFactor}, Xi::Vector{DFGVariable{ContinuousEuclid{2}}}, usrfnc::Mixture{2, Prior{Normal{Float64}}, (:c1, :c2), Tuple{DiagNormal, DiagNormal}}; multihypo::Vector{Float64}, nullhypo::Float64, threadmodel::Type, eliminated::Bool, potentialused::Bool, edgeIDs::Vector{Int64}, solveInProgress::Int64, inflation::Float64)
    @ IncrementalInference ~/.julia/dev/IncrementalInference/src/FactorGraph.jl:703
 [11] addFactor!(dfg::LightDFG{SolverParams, DFGVariable, DFGFactor}, Xi::Vector{DFGVariable{ContinuousEuclid{2}}}, usrfnc::Mixture{2, Prior{Normal{Float64}}, (:c1, :c2), Tuple{DiagNormal, DiagNormal}}; multihypo::Vector{Float64}, nullhypo::Float64, solvable::Int64, tags::Vector{Symbol}, timestamp::TimeZones.ZonedDateTime, graphinit::Bool, threadmodel::Type, suppressChecks::Bool, inflation::Float64)
    @ IncrementalInference ~/.julia/dev/IncrementalInference/src/FactorGraph.jl:1179
 [12] addFactor!(dfg::LightDFG{SolverParams, DFGVariable, DFGFactor}, xisyms::Vector{Symbol}, usrfnc::Mixture{2, Prior{Normal{Float64}}, (:c1, :c2), Tuple{DiagNormal, DiagNormal}}; multihypo::Vector{Float64}, nullhypo::Float64, solvable::Int64, timestamp::TimeZones.ZonedDateTime, tags::Vector{Symbol}, graphinit::Bool, threadmodel::Type, inflation::Float64, suppressChecks::Bool)
    @ IncrementalInference ~/.julia/dev/IncrementalInference/src/FactorGraph.jl:1224
 [13] addFactor!(dfg::LightDFG{SolverParams, DFGVariable, DFGFactor}, xisyms::Vector{Symbol}, usrfnc::Mixture{2, Prior{Normal{Float64}}, (:c1, :c2), Tuple{DiagNormal, DiagNormal}})
    @ IncrementalInference ~/.julia/dev/IncrementalInference/src/FactorGraph.jl:1218
 [14] top-level scope
    @ REPL[100]:1
```

Affie, 2021-05-04:
Ok, just saw it is caused by `Prior(I)`

I'm changing the issue to the question:

What is the correct way of defining the mechanics in a mixture?

dehann, 2021-05-04:
Hi, one of the questions perhaps is a mixture of beliefs or mixture of mechanics.  This far we do not support a mixture of different mechanics in the same factor yet.

So the question then about what is more sensible:
- Mixture(AbstractFactor,...)
- FactorX(Mixture(...))

We already support-
- FactorX(ManifoldKernelDensity(...))

hmm...