JuliaRobotics/IncrementalInference.jl #1243: missing solveKey=:parametric before solve or init

dehann, 2021-05-12:
Looks like a solveKey thing, related #1219 

MWE:

```julia
using IncrementalInference

fg = generateCanonicalFG_Kaess()

IIF.solveGraphParametric(fg)

ERROR: Solve key 'parametric' not found in variable 'l1'
Stacktrace:
 [1] error(s::String)
   @ Base ./error.jl:33
 [2] getVariableSolverData(dfg::LightDFG{SolverParams, DFGVariable, DFGFactor}, variablekey::Symbol, solvekey::Symbol)
   @ DistributedFactorGraphs ~/.julia/dev/DistributedFactorGraphs/src/services/DFGVariable.jl:478
 [3] solveGraphParametric(fg::LightDFG{SolverParams, DFGVariable, DFGFactor}; useCalcFactor::Bool, solvekey::Symbol, autodiff::Symbol, algorithm::Type, algorithmkwargs::Tuple{}, options::Optim.Options{Float64, Nothing})
   @ IncrementalInference ~/.julia/dev/IncrementalInference/src/ParametricUtils.jl:276
 [4] solveGraphParametric(fg::LightDFG{SolverParams, DFGVariable, DFGFactor})
   @ IncrementalInference ~/.julia/dev/IncrementalInference/src/ParametricUtils.jl:268
 [5] top-level scope
   @ REPL[3]:1
```

Similarly for

```julia
IIF.initParametricFrom!(fg)

┌ Error: Variable does not have solver data parametric
└ @ DistributedFactorGraphs ~/.julia/dev/DistributedFactorGraphs/src/services/DFGVariable.jl:325
ERROR: type Nothing has no field val
Stacktrace:
 [1] getproperty(x::Nothing, f::Symbol)
   @ Base ./Base.jl:33
 [2] initParametricFrom!(fg::LightDFG{SolverParams, DFGVariable, DFGFactor}, fromkey::Symbol; parkey::Symbol, onepoint::Bool)
   @ IncrementalInference ~/.julia/dev/IncrementalInference/src/ParametricUtils.jl:531
 [3] initParametricFrom! (repeats 2 times)
   @ ~/.julia/dev/IncrementalInference/src/ParametricUtils.jl:519 [inlined]
 [4] top-level scope
   @ REPL[4]:1

```

Affie, 2021-05-12:
Parametric is not added to `fg.solverParams.algorithms` by default. 
You can do:
`addParametricSolver!(fg; init=init)`

or use `solveGraphParametric!`

dehann, 2021-05-17:
> or use solveGraphParametric!

That is the one that is not working.

closed 2022-05-09 by Affie