JuliaRobotics/IncrementalInference.jl #1284: Bandwidth `vnd.bw` and manifolds

Affie, 2021-06-30:
`vnd.bw` should probably be on the tangent plane at the point in `vnd.val`:
- in coordinates makes the most sense for me.

It leads to errors of unmatched dimensions such as:

```julia

julia> getBelief(vnd)
ERROR: DimensionMismatch("tried to assign 200 elements to 300 destinations")
Stacktrace:
  [1] throw_setindex_mismatch(X::Vector{Float64}, I::Tuple{Int64})
    @ Base ./indices.jl:191
  [2] setindex_shape_check
    @ ./indices.jl:245 [inlined]
  [3] setindex!
    @ ./array.jl:864 [inlined]
  [4] makeBallTreeDensity(_pointsMatrix::Matrix{Float64}, _weightsMatrix::Vector{Float64}, _bwMatrix::Vector{Float64}, _type::DataType, addop::Tuple{typeof(+), typeof(+), typeof(+)}, diffop::Tuple{typeof(-), typeof(-), typeof(-)})
    @ KernelDensityEstimate ~/.julia/packages/KernelDensityEstimate/Xshli/src/BallTreeDensity01.jl:190
  [5] kde!(points::Matrix{Float64}, ks::Vector{Float64}, weights::Vector{Float64}, addop::Tuple{typeof(+), typeof(+), typeof(+)}, diffop::Tuple{typeof(-), typeof(-), typeof(-)})
    @ KernelDensityEstimate ~/.julia/packages/KernelDensityEstimate/Xshli/src/KDE01.jl:54
  [6] kde!(points::Matrix{Float64}, ks::Vector{Float64}, addop::Tuple{typeof(+), typeof(+), typeof(+)}, diffop::Tuple{typeof(-), typeof(-), typeof(-)})
    @ KernelDensityEstimate ~/.julia/packages/KernelDensityEstimate/Xshli/src/KDE01.jl:75
  [7] ManifoldKernelDensity(M::Sphere{2, ℝ}, ptsArr::Vector{Vector{Float64}}, bw::Vector{Float64})
    @ ApproxManifoldProducts ~/.julia/dev/ApproxManifoldProducts/src/Legacy.jl:104
  [8] manikde!
    @ ~/.julia/dev/ApproxManifoldProducts/src/API.jl:10 [inlined]
  [9] manikde!
    @ ~/.julia/dev/IncrementalInference/src/FGOSUtils.jl:133 [inlined]
 [10] getBelief(vnd::VariableNodeData{Sphere2, Vector{Float64}})
    @ IncrementalInference ~/.julia/dev/IncrementalInference/src/FactorGraph.jl:1417
 [11] top-level scope
    @ REPL[54]:1
```

dehann, 2021-07-04:
> vnd.bw should probably be on the tangent plane at the point in vnd.val:

Sounds good, no objections.  Just keep in mind that vnd.val stores N many points, so the tangent plane should either be at the mean, or one of the points.  I think the assumption I would follow here is that the bandwidth is local to each point's tangent space (since that is the assumption that was used in the previous Euclidean representation).  If need be we can store a bandwidth tangent-vector for each point, which would allow more nuanced function approximation, although not immediately necessary.  Good future direction to keep open though.

Affie, 2021-07-05:
> I think the assumption I would follow here is that the bandwidth is local to each point's tangent space (since that is the assumption that was used in the previous Euclidean representation).

That was my assumption also. One stored `bw` should work for Lie groups (if the bandwidths are the same on all points). If needed by the manifold, we can store a `bw` for every point at its tangent space. Or if the bandwidth is different on points.

> tangent plane should either be at the mean, or one of the points.

I don't think this is general enough. 

I'll update the first post with "bandwidth is local to each point's tangent space"

Affie, 2021-07-05:
It looks like the dimension issue is fixed and we are on the same page on how `bw` looks. So I'm closing this.

closed 2021-07-05 by Affie