JuliaRobotics/IncrementalInference.jl #1287: tag v0.24.5

dehann, 2021-07-01:
@JuliaRegistrator register branch=release/v0.24

JuliaRegistrator, 2021-07-01:
Registration pull request created: [JuliaRegistries/General/39989](https://github.com/JuliaRegistries/General/pull/39989)

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if the [Julia TagBot GitHub Action](https://github.com/marketplace/actions/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v0.24.5 -m "<description of version>" 7283a450ee8484f3adcba72dcb7470ea7a4c9c5b
git push origin v0.24.5
```

dehann, 2021-07-01:
@JuliaRegistrator register branch=release/v0.24

JuliaRegistrator, 2021-07-01:
Registration pull request updated: [JuliaRegistries/General/39989](https://github.com/JuliaRegistries/General/pull/39989)

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if the [Julia TagBot GitHub Action](https://github.com/marketplace/actions/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v0.24.5 -m "<description of version>" c04b30bcff092c8b10af08e900a666db014c152e
git push origin v0.24.5
```

closed 2021-07-01 by dehann