JuliaRobotics/IncrementalInference.jl #1293: Basic on-manifold example with `Sphere(2)` and one prior fail

Affie, 2021-07-05:
I've copied the example out to a test:

https://github.com/JuliaRobotics/IncrementalInference.jl/blob/23690353258509e79eb864b1b4e975750e9bfe99/test/testSphereMani.jl#L1-L32

The same example previously errored as in: #1284

dehann, 2021-07-05:
I think this part will definitely cause problems:
```julia
Base.convert(::Type{<:Tuple}, M::Sphere{2, ℝ}) = (:Euclid, :Euclid, :Euclid) 
Base.convert(::Type{<:Tuple}, ::IIF.InstanceType{Sphere{2, ℝ}})  = (:Euclid, :Euclid, :Euclid) 
```

Also not sure if this mean is working properly, since the manifold information is lost, and these are definitely not a standard Euclidean mean:
```julia
@test all(isapprox.(mean(vnd.val), [1,0,0], atol=0.1)) 
```

This mean is something much close to:
https://manoptjl.org/stable/tutorials/MeanAndMedian.html#The-given-Dataset-1

Affie, 2021-07-05:
I fixed the 3 x Euclid in the next commit of that branch. It should just be 2x. 
I’ll update the first post

Affie, 2021-07-05:
> Also not sure if this mean is working properly, since the manifold information is lost, and these are definitely not a standard Euclidean mean

The covariance is very small, so it will be very close to 1.0 and the tolerance is 0.1.

Affie, 2021-07-05:
I just remembered you said AMP is still based on the identity element. So Sphere(2) won't work for now.

Affie, 2021-07-07:
https://github.com/JuliaRobotics/ApproxManifoldProducts.jl/pull/94 fixes the dimension error.

closed 2021-07-07 by Affie