JuliaRobotics/IncrementalInference.jl #1336: `cov` function for manififolds

Affie, 2021-08-04:
Where would the correct place be for a covariance function such as this:
```julia
function Statistics.cov(vartype::InferenceVariable, ptsArr::Vector{P}) where P
  M = getManifold(vartype)
  μ = mean(M, ptsArr)
  Xcs = vee.(Ref(M), Ref(μ), log.(Ref(M), Ref(μ), ptsArr))
  return mean(Xcs .* transpose.(Xcs))
end
```
I don't know if it is general enough, but I only use it for tests currently and it works well enough.
@mateuszbaran Eventually, it should maybe be upstreamed to Manifolds? But replacing `vartype` with `M::AbstractManifold` as a parameter and making it more general.

mateuszbaran, 2021-08-04:
I think we could have it in Manifolds.jl, we already have mean, median and var. This seems simple enough, I'll try making a PR.

Also note that we can use any covariance estimator on the tangent plane, which is useful if you don't have enough points.

dehann, 2021-08-14:
Hi @mateuszbaran , @Affie , thank you very much for this -- I'll be using this function a lot in the near future!

closed 2021-08-14 by dehann