JuliaRobotics/IncrementalInference.jl #1386: Check differential factors use correct approxDeconv result

dehann, 2021-09-13:
- approxDeconv recently changed to return tangents, previously was coordinates.
- differential factors based on MKD should be built with manifold points.

Affie, 2021-09-13:
It sounds like you are on the right track, I previously tracked the issue to approxDeconv and fixed it here https://github.com/JuliaRobotics/IncrementalInference.jl/pull/1380, but I didn't have time to dig deeper.

Just note that the `sampleTangent` function should be able to handle the correct sampling already. So if MKD is built correctly it should work.

dehann, 2021-09-13:
I think the bug is happening here:
https://github.com/JuliaRobotics/IncrementalInference.jl/blob/145774ff55a452899f0d74ec372244ad0398eb1a/src/TreeMessageUtils.jl#L314-L316

Affie, 2021-09-13:
Here is the test code I have so already, but it's  focused on approxDeconv:

https://github.com/JuliaRobotics/IncrementalInference.jl/blob/145774ff55a452899f0d74ec372244ad0398eb1a/test/testSpecialEuclidean2Mani.jl#L189-L232

Affie, 2021-09-13:
> I think the bug is happening here

Yes, that needs to be converted to points. The current assumption is tangent on lie group so i think we can just use exp map at identity for now?
```julia
X = approxDeconv(tfg, afc.label, solveKey)
pts = exp(M, ident, X)
```

dehann, 2021-09-13:
Cool thanks, yes let me try that...

dehann, 2021-09-13:
and getting the following error (stable/good on left -- current master/bad on right):
[image]

PR #1387 (2021-09-13): differential factors from deconv tangents to points
a change that came as part of Manifolds.jl upgrade.  This PR restricts to only Lie group manifolds, but advances differential factors to use new approxDeconv which now returns tangents (previously was always assumed just to be tangents).  Relaxing the Lie group assumption should happen along with #1010 

Closes #1386
commit 8e929125 (2021-09-13): deconv now tangents, fix #1386
closed 2021-09-13 by dehann