JuliaRobotics/IncrementalInference.jl #1420: Refactor Mixture as direct Density

dehann, 2021-10-15:
Another question here is whether we should just update Mixture to be a regular `<:SamplableBelief` density rather than quasi factor-density structure it is now.  i.e. change everything so that `PriorPose2(Mixture((.., ...), (0.5;0.5)))`.  Should never have gone down the road of `Mixture(PriorPose2, (..,...),(0.5,0.5))` as it is now.

_Originally posted by @dehann in https://github.com/JuliaRobotics/IncrementalInference.jl/issues/1407#issuecomment-944361944_

dehann, 2022-07-15:
From later duplicate:

### Project Epic board

https://github.com/orgs/JuliaRobotics/projects/5/views/1

### Discussed in https://github.com/JuliaRobotics/Caesar.jl/discussions/808

Continue chatting on Discussion page, this Issue is just boiler plate link to Project Epics.  
( we can unlock this conversation if necessary ).

Affie, 2025-07-03:
We can also consider using Distributions.jl by expanding MixtureModel if we implement `Sampleable`.
https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/issues/1146#issuecomment-3033245065

PR #1882 (2025-09-15): Refactor Mixture as direct Density and standardize sampling on FluxModelsDistributions to resemble <: Sampleable
Needs: https://github.com/JuliaRobotics/ApproxManifoldProducts.jl/pull/320

Close #1420

Note: 
- Sampleable beliefs need to work the same as <: Sampleable
- Only somewhat breaking, but mostly kept the Mixture signature.