JuliaRobotics/IncrementalInference.jl #1478: Update local instances to DefaultDFG

dehann, 2022-02-01:
With DFG v0.18.0, probably a good idea to standardize the names better.  Remove all references to LightDFG.

Affie, 2022-02-02:
So we currently have an IIF defined:
`InMemDFGType`
and now the new DFG defined:
`LocalDFG`

Originally IIF's `InMemDFGType` was used to avoid breaking changes and ease the transition from the original graphs to light graphs.

For me `LocalDFG` now means the default in-memory DFG type to use.

So do we merge the two names to only use DFG's `LocalDFG`?
It should be stable enough to not cause future issues.

Otherwise, we can have  `const InMemDFGType=LocalDFG` in IIF.

@dehann , let me know your thoughts and if you understand the above the same.

dehann, 2022-02-03:
I think lets merge, only have `LocalDFG`.  Thanks for taking this one!

PR #1479 (2022-02-04): Standardise to LocalDFG in IIF
Close #1478
closed 2022-02-04 by Affie