JuliaRobotics/IncrementalInference.jl #1480: CalcFactor.cache in Parametric case

dehann, 2022-02-10:
Users can now overload:
```julia
IIF.preambleCache(dfg::AbstractDFG, vars::AbstractVector{Symbol}, usrfnc::AbstractFactor)
```

which should* runs once at factor creation.  Cache is stored at `ccw.dummyCache::C` and `CalcFactor.cache::C`.

Parametric code needs to be refactored to allow `preampleCache(dfg, vars, usrfnc)` function to run here:

Affie, 2022-03-29:
This breaks a lot of things in parametric. I'll need some help understanding this.

dehann, 2022-03-31:
Okay, we can do a worksession.  First week in April?

dehann, 2022-06-30:
Hi Johan, sorry I don't quite follow your question?  I think you mean what is the signature for `preambleCache` (see below)

So default for a factor is `cache::Nothing`.  That gets put into `CalcFactor.cache::Nothing`.  So `getSample` and residual functions will just see `cf.cache === nothing`.

If one goes and overrides `preambleCache(dfg::AbstractDFG, vars::AbstractVector{<:DFGVariable}, fct::MyFactor)` and returns an object, say type `::T`.  Then once the thing is going, the `getSample` and residual functions will see that as `cf.cache::T`, and the user can put whatever they want in the cache (links to YouTube if it so turns out).

Here are the docs for `preambleCache`:
- https://juliarobotics.org/Caesar.jl/latest/concepts/stash_and_cache/#IncrementalInference.preambleCache

I tried to capture most of the headline stuff in docs, perhaps that helps if above does not answer yet?
- https://juliarobotics.org/Caesar.jl/latest/concepts/stash_and_cache/

Happy to answer more?

Affie, 2022-06-30:
The second parameter here: `IIF.preambleCache(dfg::AbstractDFG, vars::AbstractVector{<:DFGVariable}, fct::MyFactor)` is not consistent in source and docs.  
It is the vector of symbols in some places (including docs) `AbstractVector{Symbol}`

Affie, 2022-06-30:
From doc used as symbols:
`addFactor!(fg, [:a;:b], mfc)`

Affie, 2022-06-30:
Sorry copied wrong line, see RoME in places:
`preambleCache(::AbstractDFG, ::AbstractVector{Symbol}, ::Pose2Point2Bearing) = P2P2BearingReuse()`
I'm guessing it's just a bug then?

dehann, 2022-06-30:
oh sorry, let's fix that

dehann, 2022-06-30:
is that better?

dehann, 2022-07-01:
Oh, that is a bug yes.  Should be `::AbstractVector<:DFGVariable}`

Affie, 2022-07-04:
It looks like for simple factors such as `Pose2Pose2` the cache doesn't make a real difference, here is Pose2Pose2 without a cache:
[image]

dehann, 2022-07-05:
My sense is to get performance, we need to get allocations down to 0.  The only way to do that is with in-place operations everywhere.  The only way we can do in-place is to have place to preallocate memory, and the only way I see to do that is with `cf.cache`?

Affie, 2022-07-05:
Currently, we cannot get allocations down to 0, this example uses SArrays for p,q and MArray for X_hat
```
julia> @benchmark log!(M, X_hat, p, q)
BenchmarkTools.Trial: 10000 samples with 983 evaluations.
 Range (min … max):   61.297 ns … 38.177 μs  ┊ GC (min … max):  0.00% … 99.67%
 Time  (median):      66.344 ns              ┊ GC (median):     0.00%
 Time  (mean ± σ):   110.464 ns ±  1.131 μs  ┊ GC (mean ± σ):  30.69% ±  2.99%

  ▆██▇▆▅▄▃▂▁▁▁▁▁▁                    ▂▃▂▄▅▂                    ▂
  ██████████████████▇▇▆▆▆▆▇█▇▇█▆▆▆▆▆▆███████▇▇▇▆▅▆▆▆▃▅▅▃▃▅▄▃▃▄ █
  61.3 ns       Histogram: log(frequency) by time       157 ns <

 Memory estimate: 192 bytes, allocs estimate: 4.
```

Affie, 2022-07-05:
I might have a workaround for my parametric issues:
```
  calcf = CalcFactor(getFactorMechanics(cf), _getFMdThread(fct), 0, 0, (), [], true, cache)
  # 0.231179 seconds (705.03 k allocations: 148.488 MiB, 23.06% gc time)
  calcf = CalcFactor(IIF.getFactorMechanics(cf), nothing, 0, 0, nothing, nothing, true, nothing)
  # 0.000788 seconds (2.44 k allocations: 411.891 KiB)
```

dehann, 2022-07-05:
> we cannot get allocations down to 0, this example uses SArrays for p,q and MArray for X_hat

Right, so first cut is then to get everything on the stack and avoid allocations on the heap?

> Also, remember M = getManifold(Pose2) for example uses constant propagation so it's faster not to fetch out of the cache (as far as I understand).

Okay, I'm not knowledgeable enough to easily answer that, but sounds like it can be faster that way.

> the only way I see to do that is with `cf.cache`?

oops, there is another way i forgot about.  The factor struct itself can hold memory or fields, but user needs to be careful on how to use multithreading.  Part of `cf.cache` was that IIF can manage more of the multithreading memory.  Separate story.

Another thing, I was thinking we should get `cf` / `cf.cache` completely allocated on the stack which would also make it faster.

Affie, 2022-07-05:
> Right, so first cut is then to get everything on the stack and avoid allocations on the heap?

Matheusz is updating some manifolds to work faster with SArray, so I think it should be on the stack then. Parametric is not a big lift and I tried it as an experiment already.

I think we should consider upgrading IIF to be able to work with SArrays before we do #1010?

dehann, 2022-07-05:
> I think we should consider upgrading IIF to be able to work with SArrays before we do https://github.com/JuliaRobotics/IncrementalInference.jl/issues/1010?

That's a good idea, we should be able to do that relatively quickly.

> Matheusz is updating some manifolds to work faster with SArray, so I think it should be on the stack then. Parametric is not a big lift and I tried it as an experiment already.

Ah that's great, sounds good

closed 2022-08-09 by Affie