JuliaRobotics/IncrementalInference.jl #1498: Unable to pack multihypo

GearsAD, 2022-03-06:
Still need to debug, but it looks like Rayleigh is failing to pack:

```
Error while packing 'x1x2f1' as 'Mixture{2, LinearRelative, (:hypo1, :hypo2), Tuple{Rayleigh{Float64}, Uniform{Float64}}}(LinearRelative{1, FullNormal}(FullNormal(
dim: 1
μ: [0.0]
Σ: [1.0;;]
)
), (hypo1 = Rayleigh{Float64}(σ=3.0), hypo2 = Uniform{Float64}(a=30.0, b=55.0)), Categorical{Float64, Vector{Float64}}(support=Base.OneTo(2), p=[0.4, 0.6]), 1, [2])', please check the unpacking/packing converters for this factor - 
MethodError: Cannot `convert` an object of type Rayleigh{Float64} to an object of type PackedSamplableBelief
Closest candidates are:
  convert(::Type{<:PackedSamplableBelief}, !Matched::Type{<:ManifoldKernelDensity}) at /srv/julia/pkg/packages/IncrementalInference/El6vc/src/Serialization/services/SerializationMKD.jl:18
  convert(::Type{<:PackedSamplableBelief}, !Matched::NamedTuple) at /srv/julia/pkg/packages/IncrementalInference/El6vc/src/Serialization/services/SerializingDistributions.jl:122
  convert(::Type{<:PackedSamplableBelief}, !Matched::Union{AliasingScalarSampler, BallTreeDensity, DiscreteNonParametric, MvNormal, Normal, Uniform, HeatmapGridDensity, LevelSetGridNormal, MKD{M, B} where {M<:AbstractManifold, B<:BallTreeDensity}}) at /srv/julia/pkg/packages/IncrementalInference/El6vc/src/Serialization/services/SerializingDistributions.jl:119
  ...
Stacktrace:
  [1] convert(#unused#::Type{PackedMixture}, obj::Mixture{2, LinearRelative, (:hypo1, :hypo2), Tuple{Rayleigh{Float64}, Uniform{Float64}}})
    @ IncrementalInference /srv/julia/pkg/packages/IncrementalInference/El6vc/src/Factors/Mixture.jl:122
  [2] convert(#unused#::Type{PackedFunctionNodeData{PackedMixture}}, d::FunctionNodeData{CommonConvWrapper{Mixture{2, LinearRelative, (:hypo1, :hypo2), Tuple{Rayleigh{Float64}, Uniform{Float64}}}, Nothing, Vector{Int64}, NamedTuple{(:x1, :x2), Tuple{Vector{Any}, Vector{Any}}}, Nothing, Vector{Float64}, Nothing}})
    @ IncrementalInference /srv/julia/pkg/packages/IncrementalInference/El6vc/src/Serialization/services/DispatchPackedConversions.jl:8
  [3] _packSolverData(f::DFGFactor{CommonConvWrapper{Mixture{2, LinearRelative, (:hypo1, :hypo2), Tuple{Rayleigh{Float64}, Uniform{Float64}}}, Nothing, Vector{Int64}, NamedTuple{(:x1, :x2), Tuple{Vector{Any}, Vector{Any}}}, Nothing, Vector{Float64}, Nothing}, 2}, fnctype::Mixture{2, LinearRelative, (:hypo1, :hypo2), Tuple{Rayleigh{Float64}, Uniform{Float64}}}; base64Encode::Bool)
    @ DistributedFactorGraphs /srv/julia/pkg/packages/DistributedFactorGraphs/ttnFL/src/services/Serialization.jl:301
  [4] _packSolverData
    @ /srv/julia/pkg/packages/DistributedFactorGraphs/ttnFL/src/services/Serialization.jl:299 [inlined]
  [5] packFactor(dfg::GraphsDFG{SolverParams, DFGVariable, DFGFactor}, f::DFGFactor{CommonConvWrapper{Mixture{2, LinearRelative, (:hypo1, :hypo2), Tuple{Rayleigh{Float64}, Uniform{Float64}}}, Nothing, Vector{Int64}, NamedTuple{(:x1, :x2), Tuple{Vector{Any}, Vector{Any}}}, Nothing, Vector{Float64}, Nothing}, 2})
    @ DistributedFactorGraphs /srv/julia/pkg/packages/DistributedFactorGraphs/ttnFL/src/services/Serialization.jl:328
  [6] top-level scope
    @ In[21]:4
  [7] eval
    @ ./boot.jl:373 [inlined]
  [8] include_string(mapexpr::typeof(REPL.softscope), mod::Module, code::String, filename::String)
    @ Base ./loading.jl:1196
  [9] softscope_include_string(m::Module, code::String, filename::String)
    @ SoftGlobalScope /srv/julia/pkg/packages/SoftGlobalScope/u4UzH/src/SoftGlobalScope.jl:65
 [10] execute_request(socket::ZMQ.Socket, msg::IJulia.Msg)
    @ IJulia /srv/julia/pkg/packages/IJulia/e8kqU/src/execute_request.jl:67
 [11] #invokelatest#2
    @ ./essentials.jl:716 [inlined]
 [12] invokelatest
    @ ./essentials.jl:714 [inlined]
 [13] eventloop(socket::ZMQ.Socket)
    @ IJulia /srv/julia/pkg/packages/IJulia/e8kqU/src/eventloop.jl:8
 [14] (::IJulia.var"#15#18")()
    @ IJulia ./task.jl:423

Stacktrace:
 [1] error(s::String)
   @ Base ./error.jl:33
 [2] _packSolverData(f::DFGFactor{CommonConvWrapper{Mixture{2, LinearRelative, (:hypo1, :hypo2), Tuple{Rayleigh{Float64}, Uniform{Float64}}}, Nothing, Vector{Int64}, NamedTuple{(:x1, :x2), Tuple{Vector{Any}, Vector{Any}}}, Nothing, Vector{Float64}, Nothing}, 2}, fnctype::Mixture{2, LinearRelative, (:hypo1, :hypo2), Tuple{Rayleigh{Float64}, Uniform{Float64}}}; base64Encode::Bool)
   @ DistributedFactorGraphs /srv/julia/pkg/packages/DistributedFactorGraphs/ttnFL/src/services/Serialization.jl:314
 [3] _packSolverData
   @ /srv/julia/pkg/packages/DistributedFactorGraphs/ttnFL/src/services/Serialization.jl:299 [inlined]
 [4] packFactor(dfg::GraphsDFG{SolverParams, DFGVariable, DFGFactor}, f::DFGFactor{CommonConvWrapper{Mixture{2, LinearRelative, (:hypo1, :hypo2), Tuple{Rayleigh{Float64}, Uniform{Float64}}}, Nothing, Vector{Int64}, NamedTuple{(:x1, :x2), Tuple{Vector{Any}, Vector{Any}}}, Nothing, Vector{Float64}, Nothing}, 2})
   @ DistributedFactorGraphs /srv/julia/pkg/packages/DistributedFactorGraphs/ttnFL/src/services/Serialization.jl:328
 [5] top-level scope
   @ In[21]:4
 [6] eval
   @ ./boot.jl:373 [inlined]
 [7] include_string(mapexpr::typeof(REPL.softscope), mod::Module, code::String, filename::String)
   @ Base ./loading.jl:1196
```

dehann, 2022-03-06:
Rayleigh may already be fixed here:
- #1496

GearsAD, 2022-03-06:
Trying :eyes:

GearsAD, 2022-03-06:
Still failing unfortunately.

Here's the MRE:
```julia
using IncrementalInference
using DistributedFactorGraphs
using JSON

# Start with an empty factor graph
fg = initfg()

# add the first node
addVariable!(fg, :x0, ContinuousScalar)
addVariable!(fg, :x1, ContinuousScalar)
mmo = Mixture(LinearRelative, 
              (hypo1=Rayleigh(3), hypo2=Uniform(30,55)), 
              [0.4; 0.6])
addFactor!(fg, [:x0, :x1], mmo)
JSON.json(packFactor(fg, getFactor(fg, :x0x1f1)))
```

dehann, 2022-03-06:
Alright, I'll have a look.  Just having lunch

dehann, 2022-03-06:
Looks like its a really silly fix, 93abed1.  I'm busy tagging v0.27.3 now.

PR #1501 (2022-03-06): test and expand serializable beliefs with Rayleigh
fix #1498
closed 2022-03-06 by dehann