JuliaRobotics/IncrementalInference.jl #1500: Prior standardization

GearsAD, 2022-03-06:
As a user I'd like priors to be standardized across the various dimensions, either one prior for all (like Prior?) or a set of priors that are named the same.

dehann, 2022-03-06:
`Prior` will not be able to be used as a standard.  There is a newly created `ManifoldPrior` which would be better but is clunky.  The preferred approach for ease of use is to use names:
- `PriorPosition1`, `PriorPosition2`...
- `PriorCircular`
- `PriorPose2`, `PriorPose3`, ...
- `PriorInertialPose3`,
- ...

GearsAD, 2022-03-06:
Great, thanks - keep me in the loop if there change and I'll make sure we accommodate the new factors in the API. Aliases are great then we can deprecate the older ones over time.