JuliaRobotics/IncrementalInference.jl #1503: tag v0.27.3

dehann, 2022-03-06:
@JuliaRegistrator regsiter branch=release/v0.27

JuliaRegistrator, 2022-03-06:
Error while trying to register: Action not recognized: regsiter

dehann, 2022-03-06:
@JuliaRegistrator register branch=release/v0.27

JuliaRegistrator, 2022-03-06:
Registration pull request created: [JuliaRegistries/General/56093](https://github.com/JuliaRegistries/General/pull/56093)

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if the [Julia TagBot GitHub Action](https://github.com/marketplace/actions/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v0.27.3 -m "<description of version>" 6eaf5c9e4d02ee07c646f41b8981dba32a6e0125
git push origin v0.27.3
```

closed 2022-03-06 by dehann