JuliaRobotics/IncrementalInference.jl #1527: Consider using `identity_like_element`

Affie, 2022-05-30:
From a suggestion of @mateuszbaran 
> "You don't have to restrict to group manifolds. Some other have a decent "identity-like" thing you can use as a default."

for example
```julia
identity_like_element(M::AbstractDecoratorManifold) = identity_element(M)
identity_like_element(M::Euclidean) = zeros(representation_size(M))

function samplePoint(M::AbstractManifold, sbelief, p=identity_like_element(M); retraction_method::AbstractRetractionMethod=default_retraction_method(M))
  X = sampleTangent(M, sbelief, p)
  return retract(M, p, X, retraction_method)
end
```

@dehann, I don't know if it will still be applicable after the https://github.com/JuliaRobotics/ApproxManifoldProducts.jl/issues/41 upgrade, but it looks like a step in the right direction.

dehann, 2022-05-31:
Yes we definitely need to do that.  I've been using the user provided 'getPointDefault' for the same
purpose.  

Agreed, as we do AMP 41, we should also move away from the group restriction to more general Riemannian

kellertuer, 2022-06-01:
I am not so sure, since a reasonable choice can probably only be made on Hadamard manifolds?
For example For `Euclidean` (as above) or `SymetricPOsitiveDefinite` (identity matrix) this should work fine, I still do not like the name then though.

* I am unsure what this function should provide (the `identity_like_thingy`
* What is `sample_point` supposed to do (actually I could prefer underscore names for functions), for example what is `sbelief`?
* On the sphere I would for example not be able to provide such a point, since if – for example – the South Pole is on your data, you do not want to have the North Pole as such a point.

mateuszbaran, 2022-06-01:
> I am not so sure, since a reasonable choice can probably only be made on Hadamard manifolds?

Yes, I don't have a good example of an "identity-like element" for manifolds that are not either a Lie group or a Hadamard manifold (or their product).

> * On the sphere I would for example not be able to provide such a point, since if – for example – the South Pole is on your data, you do not want to have the North Pole as such a point.

I think it would be fine to just not implement it on some manifolds like sphere.

kellertuer, 2022-06-01:
and why would we call it then `identity_like` – what makes it alike the identity?
For Euclidean (w.r.t. the addition group) it would actually be the identity. for SPDs (w.r.t. the matrix multiplication group) it would also be the identity; so then I would mainly wonder about the name.
Maybe similar to the metric (and a default metric) we could implement `identity` for these manifolds having a default (Lie) group in mind; though for SPDs matrix multiplication is not Lie I think?

mateuszbaran, 2022-06-01:
SPDs can be equipped with a group structure though not under matrix multiplication: https://cpb-us-w2.wpmucdn.com/blog.nus.edu.sg/dist/f/11899/files/2019/12/M122108.pdf .

So perhaps we could just extend `identity_element` to Euclidean and SPDs, Rotations and maybe something else and it would be fine?

dehann, 2022-06-02:
Yeah, I was not sure either exactly what to do for "identity" for more general manifolds.  Basically came up with two answers for our use case:
- User knows and specifies a "default" point when defining a 'VariableType' in our system.  This happens at the same time as when the default representation for a point is specified by a user, e.g. https://github.com/JuliaRobotics/RoME.jl/blob/87718625fe4ce1d039814615b7f3acdaf03eb5e8/src/variables/VariableTypes.jl#L35 
- During calculations, we are able to generate tangent spaces at points based on available information.  E.g. the mean of a collection of sample points can be used in some instances during calculation, but that only works in specific cases during computation somewhere and the user of the system will likely never see that.

If there are "default" points available upstream from Manifolds.jl, we are very likely to just use that, or fall back on a user definition somehow.

Affie, 2022-06-03:
> What is sample_point supposed to do (actually I could prefer underscore names for functions), for example what is sbelief?

`sbelief` is a samplable belief, it includes distributions from Distributions.jl, BallTreeDensity from KDE, ManifoldKernelDensity, FluxModelsDistribution, and so on. 

The samplable beliefs are defined in euclidean and `samplePoint` just creates a point on the relevant manifold.

mateuszbaran, 2022-06-03:
> If there are "default" points available upstream from Manifolds.jl, we are very likely to just use that, or fall back on a user definition somehow.

If you need a "default" point on a sphere for example then I think you should have your own function for it. We will only extend `identity_element` to manifolds which have a clear candidate.

Affie, 2022-06-28:
We are not going to be using `identity_like_element`, but rather a Manifolds.jl defined `identity_element` on a per need basis on manifolds that have a clear candidate. We can evaluate the needs later when we do https://github.com/JuliaRobotics/ApproxManifoldProducts.jl/issues/41

closed 2022-06-28 by Affie

dehann, 2022-06-28:
AMP41 will likely now adopt the Manifolds.jl defaults, and should definitely not restrict to groups.  AMP41 should go for Riemannian