JuliaRobotics/IncrementalInference.jl #1530: MethodError: no method matching deepcopy_internal on julia 1.8 rc1

Affie, 2022-06-02:
In Test solveCliqueUp! and solveCliqDown! at ../test/testCliqSolveDbgUtils.jl:6

```julia
MethodError: no method matching deepcopy_internal(::ReentrantLock)
Closest candidates are:
  deepcopy_internal(::Base.AbstractLock, ::IdDict) at deepcopy.jl:130
  deepcopy_internal(::Any, ::IdDict) at deepcopy.jl:53
  deepcopy_internal(::BigInt, ::IdDict) at gmp.jl:794
  ...
Stacktrace:
  [1] deepcopy_internal(x::Base.GenericCondition{ReentrantLock}, stackdict::IdDict{Any, Any})
    @ Base ./deepcopy.jl:143
  [2] deepcopy_internal(x::Any, stackdict::IdDict{Any, Any})
    @ Base ./deepcopy.jl:65
  [3] deepcopy_internal(x::Dict{Symbol, Any}, stackdict::IdDict{Any, Any}) (repeats 2 times)
    @ Base ./deepcopy.jl:125
  [4] deepcopy_internal(x::Any, stackdict::IdDict{Any, Any}) (repeats 3 times)
    @ Base ./deepcopy.jl:65
  [5] deepcopy
    @ ./deepcopy.jl:26 [inlined]
  [6] (::StateMachine{CliqStateMachineContainer})(userdata::CliqStateMachineContainer{BayesTreeNodeData, GraphsDFG{SolverParams, DFGVariable, DFGFactor}, GraphsDFG{SolverParams, DFGVariable, DFGFactor}, MetaBayesTree}, timeout::Nothing; pollinterval::Float64, breakafter::Function, verbose::Bool, verbosefid::Base.TTY, verboseXtra::IncrementalInference.CliqStatus, iterlimit::Int64, injectDelayBefore::Nothing, recordhistory::Bool, housekeeping_cb::IncrementalInference.var"#479#481"{IncrementalInference.TreeClique, CliqStateMachineContainer{BayesTreeNodeData, GraphsDFG{SolverParams, DFGVariable, DFGFactor}, GraphsDFG{SolverParams, DFGVariable, DFGFactor}, MetaBayesTree}})
    @ FunctionalStateMachine ~/.julia/packages/FunctionalStateMachine/2JZFG/src/StateMachine.jl:76
  [7] initStartCliqStateMachine!(dfg::GraphsDFG{SolverParams, DFGVariable, DFGFactor}, tree::MetaBayesTree, cliq::IncrementalInference.TreeClique, timeout::Nothing; oldcliqdata::BayesTreeNodeData, verbose::Bool, verbosefid::Base.TTY, drawtree::Bool, show::Bool, incremental::Bool, limititers::Int64, upsolve::Bool, downsolve::Bool, recordhistory::Bool, delay::Bool, logger::Base.CoreLogging.SimpleLogger, solve_progressbar::Nothing, algorithm::Symbol, solveKey::Symbol)
    @ IncrementalInference ~/.julia/packages/IncrementalInference/1ELXb/src/CliqueStateMachine.jl:64
  [8] tryCliqStateMachineSolve!(dfg::GraphsDFG{SolverParams, DFGVariable, DFGFactor}, treel::MetaBayesTree, cliqKey::IncrementalInference.CliqueId{Int64}, timeout::Nothing; oldtree::MetaBayesTree, verbose::Bool, verbosefid::Base.TTY, drawtree::Bool, limititers::Int64, downsolve::Bool, incremental::Bool, delaycliqs::Vector{Symbol}, recordcliqs::Vector{Symbol}, solve_progressbar::Nothing, algorithm::Symbol, solveKey::Symbol)
    @ IncrementalInference ~/.julia/packages/IncrementalInference/1ELXb/src/SolverAPI.jl:112
  [9] solveCliqUp!(fg::GraphsDFG{SolverParams, DFGVariable, DFGFactor}, tree::MetaBayesTree, cliqid::Int64, solveKey::Symbol, beliefMessages::Dict{Int64, LikelihoodMessage}; verbose::Bool, recordcliq::Bool)
    @ IncrementalInference ~/.julia/packages/IncrementalInference/1ELXb/src/SolverAPI.jl:483
 [10] macro expansion
    @ ~/.julia/packages/IncrementalInference/1ELXb/test/testCliqSolveDbgUtils.jl:32 [inlined]
 [11] macro expansion
    @ ~/.julia/juliaup/julia-1.8.0-rc1+0~x64/share/julia/stdlib/v1.8/Test/src/Test.jl:1357 [inlined]
 [12] top-level scope
    @ ~/.julia/packages/IncrementalInference/1ELXb/test/testCliqSolveDbgUtils.jl:10
 [13] include(fname::String)
    @ Base.MainInclude ./client.jl:476
 [14] top-level scope
    @ ~/.julia/packages/IncrementalInference/1ELXb/test/runtests.jl:14
 [15] include(fname::String)
    @ Base.MainInclude ./client.jl:476
 [16] top-level scope
    @ none:6
 [17] eval
    @ ./boot.jl:368 [inlined]
 [18] exec_options(opts::Base.JLOptions)
    @ Base ./client.jl:276
 [19] _start()
    @ Base ./client.jl:522┌ Warning: printCliqHistorySummary -- No CSM history found.
└ @ IncrementalInference ~/.julia/packages/IncrementalInference/1ELXb/src/TreeDebugTools.jl:191
```

dehann, 2022-06-02:
Looks like we should create a new lock or something similar on this line in FSM:
https://github.com/JuliaRobotics/FunctionalStateMachine.jl/blob/a534cb2bf1cd0571c5422034f8febca1b38fe4b9/src/StateMachine.jl#L76

Affie, 2022-08-19:
It looks like it will be fixed here: https://github.com/JuliaLang/julia/pull/46406

Affie, 2022-08-19:
In the meantime here is a fix that just does not copy the channels that are causing the issue: https://github.com/JuliaRobotics/IncrementalInference.jl/pull/1601

closed 2022-08-19 by Affie