JuliaRobotics/IncrementalInference.jl #1537: Standardize ArrayPartition, Manifolds.ProductRepr, and Tuple[...]

dehann, 2022-06-19:
At time of writing Manifolds is slowly working to towards just `ArrayPartition`, but it's clear cut yet.  So code current supports mostly `ProductRepr`.

Recently we did a workaround for `initVariable!` to also accept `Tuple[...]` as an input type, see:
- #1535

Affie, 2022-07-05:
See a related issue in Manifolds https://github.com/JuliaManifolds/Manifolds.jl/issues/497

Affie, 2022-07-05:
See https://github.com/JuliaRobotics/IncrementalInference.jl/issues/1546#issuecomment-1173605410 for `ArrayPartition` vs `ProductRepr`

Affie, 2022-07-08:
See https://github.com/JuliaRobotics/IncrementalInference.jl/pull/1552 to allow support for `ArrayPartition`. 
Both can be supported, but mixing will be hard and increase code maintanaince. We are working towards using `ArrayPartition`

Affie, 2022-07-08:
ArrayPartition makes a few things easier, here is a snippet of how it works:
```juila
julia> a = ArrayPartition([1.,2], [3 5; 4 6])
([1.0, 2.0], [3 5; 4 6])

julia> a[1]
1.0

julia> a[2]
2.0

julia> a[3]
3

julia> a[:]
6-element Vector{Float64}:
 1.0
 2.0
 3.0
 4.0
 5.0
 6.0
```

Affie, 2022-07-11:
`ArrayPartition` is the standard from v0.30 forward.

closed 2022-07-11 by Affie

dehann, 2022-07-11:
very cool thanks!