JuliaRobotics/IncrementalInference.jl #1564: A new method for CCW is compiled on every factor added

Affie, 2022-07-15:
It actually looks like these 2 CCW functions gets 2 new methods each. 
``` julia
Symbol("#CommonConvWrapper#234")
:CommonConvWrapper
```
Tested with this and used MethodAnalysis to track down to CCW.
```julia

using IncrementalInference

N=2 # run once and then set to 3,4 etc and run again
fg = initfg()
fg.solverParams.graphinit = false
addVariable!(fg, :x0, ContinuousScalar)
addFactor!(fg, [:x0], Prior(Normal(0.0,0.1)))
dist = Normal(1.0,0.1)

for i=1:N
    addVariable!(fg, Symbol("x$i"), ContinuousScalar)
    @time addFactor!(fg, [Symbol("x$(i-1)"),Symbol("x$i")], LinearRelative(dist))
end
```

Affie, 2022-07-15:
Tracked it down: it is the `NamedTuple` in CCW

dehann, 2022-07-15:
wow

dehann, 2022-07-15:
Oh, btw, think the same thing is happening with multihypo recipes:
https://github.com/JuliaRobotics/IncrementalInference.jl/blob/40962b76cc03c55b49bfded871a7db60f0a0fd22/src/ExplicitDiscreteMarginalizations.jl#L227

Affie, 2022-07-15:
We should also check Mixture as part of the rework there:

https://github.com/JuliaRobotics/IncrementalInference.jl/blob/40962b76cc03c55b49bfded871a7db60f0a0fd22/src/Factors/Mixture.jl#L40-L48

closed 2022-07-15 by Affie

dehann, 2022-07-15:
Yeah Mixture should be much simplified:
- #1420

PR #1565 (2022-07-15): CCW.params NamedTuple->Tuple
Close #1564
Close #1513

Also cleanup:
```julia
# FIXME deprecate use of (:null,)
  mani = length(Xi)==0 || sfidx==0 ? (:null,) : getManifold(Xi[sfidx])
```