JuliaRobotics/IncrementalInference.jl #1566: Does multihypo NamedTuple recipe require dispatch?

dehann, 2022-07-15:
Oh, btw, think the same thing might be happening with multihypo recipes:
https://github.com/JuliaRobotics/IncrementalInference.jl/blob/40962b76cc03c55b49bfded871a7db60f0a0fd22/src/ExplicitDiscreteMarginalizations.jl#L227

_Originally posted by @dehann in https://github.com/JuliaRobotics/IncrementalInference.jl/issues/1564#issuecomment-1185633151_

dehann, 2022-07-15:
Probably best to just change this to a hard type.

Affie, 2022-07-15:
Should also probably fix:

https://github.com/JuliaRobotics/IncrementalInference.jl/blob/40962b76cc03c55b49bfded871a7db60f0a0fd22/src/services/EvalFactor.jl#L155-L157

Affie, 2022-07-15:
and:
https://github.com/JuliaRobotics/IncrementalInference.jl/blob/40962b76cc03c55b49bfded871a7db60f0a0fd22/src/services/EvalFactor.jl#L235-L237

Affie, 2022-07-15:
> (;certainidx, allelements, activehypo, mhidx) 

That should actually be fine.

dehann, 2022-07-15:
> Yes I agree. The bad case is if the names are variable. 

So is this fine to use with good performance?
```julia
struct MyType1{T <: NamedTuple}
  x::T
end
```

Affie, 2022-07-15:
It will be a tradeoff between compile time and runtime. Every new NamedTuple will require a new method to be compiled.

Affie, 2022-07-26:
This tip is relevant here: https://docs.julialang.org/en/v1/manual/performance-tips/#The-dangers-of-abusing-multiple-dispatch-(aka,-more-on-types-with-values-as-parameters)

dehann, 2022-07-29:
Agreed, thanks.

dehann, 2023-01-02:
Changed this to a hard type `HypoRecipe` anyway:
- #1669

closed 2023-01-02 by dehann