JuliaRobotics/IncrementalInference.jl #1570: "Basic 3 door, trinary multihypothesis tests" often fails numerically

Affie, 2022-07-18:


dehann, 2022-07-19:
This must be a regression.  It was working well.

Affie, 2022-07-26:
I checked v0.28 and it was failing intermittently there as well, so think we can safely eliminate the recent `ArrayPartition` and Manifolds v0.8 upgrade.

dehann, 2022-07-29:
What I mean is the frequency of failures seem much higher to me than I remember v0.26-28 being.  I looked into this a bit and could not immediately see anything wrong.  Perhaps the test is just too strict, or we should change the testing strategy on this file for more amenable outcomes to current performance.

Affie, 2022-08-09:
From #1580, it looks like this will only be done after #1010

dehann, 2022-08-10:
i still think this test used to be better.  I looked at it he other day and seems like it's "working" just a bit less often than before.  So there is possibly as soft bug somewhere, but let's keep moving until something more shows up