JuliaRobotics/IncrementalInference.jl #1573: TreeMsgLikelihoods should swap NamedTuple for Tuple

dehann, 2022-07-26:
Could have a similar compile time implication as was found in:
- #1565

Affie, 2022-07-26:
Are you referring to this?
https://github.com/JuliaRobotics/IncrementalInference.jl/blob/60134f968860d5b3a0de6eb429adbb08ef54f24d/src/entities/CliqueTypes.jl#L43-L56

This is good as a NamedTuple.

dehann, 2022-07-29:
Looking at `.sender` I think that should be okay, since the type is hard coded.

Not recalling exactly, but upon looking through this again I think it was actually the `NamedTuple` in the MsgRelativeType const that might be causing the Bayes tree compile time delays.

Just quickly, I made a very small runtime improvement:
- #1579 

But does notyet get rid of the potential `NamedTuple` compile time issue on tree message passing.

I'm also worried about the non-concrete union type on `.cliqueLikelihood`