JuliaRobotics/IncrementalInference.jl #1574: Regression, new bug, related to useMsgLikelihoods

dehann, 2022-07-26:
Looks like it has something to do with `getSolverParams(fg).useMsgLikelihoods = true`.  MWE:

```julia
using IncrementalInference

fg = generateGraph_CaesarRing1D()
getSolverParams(fg).useMsgLikelihoods = true

solveGraph!(fg)
```

Code completes fine if `useMsgLikelihoods=false`

_Originally posted by @dehann in https://github.com/JuliaRobotics/IncrementalInference.jl/issues/1572#issuecomment-1194719417_

dehann, 2022-07-26:
Was first seen in `testDeconv.jl`

dehann, 2022-07-26:
This is the printout:
https://github.com/JuliaRobotics/IncrementalInference.jl/blob/60134f968860d5b3a0de6eb429adbb08ef54f24d/src/Factors/LinearRelative.jl#L41

@Affie, what does the "H" mean?

dehann, 2022-07-26:
So here is one inconsistency, `sampleFactor` returns different types depending on whether the factor was created by the user or from a tree likelihood message:

This is good (returns `Vector{Flpat64}`):
https://github.com/JuliaRobotics/IncrementalInference.jl/blob/c249ff1e3bae8e554c41b2414acc4f8592f6db63/test/testUseMsgLikelihoods.jl#L16

But then just lower down this is bad (returns `Matrix{Float64}`):
https://github.com/JuliaRobotics/IncrementalInference.jl/blob/c249ff1e3bae8e554c41b2414acc4f8592f6db63/test/testUseMsgLikelihoods.jl#L63

dehann, 2022-07-26:
Ah, that's because the differential factor from the tree likelihood factor is and MKD which uses a different getSample function.  There is some inconsistency there.

dehann, 2022-07-26:
Okay so `sampleTangent` on `::Normal` returns a `::Vector`:
https://github.com/JuliaRobotics/IncrementalInference.jl/blob/b4e9c6f80732b4917a1e25cba5dc7c62dec0c005/test/testUseMsgLikelihoods.jl#L19-L20

while `sampleTangent` on `::MKD` returns a `::Matrix`:
https://github.com/JuliaRobotics/IncrementalInference.jl/blob/b4e9c6f80732b4917a1e25cba5dc7c62dec0c005/test/testUseMsgLikelihoods.jl#L83-L84

dehann, 2022-07-26:
So fixed with forced type change of sample return for MKD case.  If matrix has just one column, then convert to a vector:
71de84

Affie, 2022-07-26:
KDE returns a matrix on `sample` and it is used as coordinates which should be a vector, a recent update in Manifolds.jl showed this error.

PR #1576 (2022-07-26): Fix samplTangent MKD coords, and test
Pick some changes from #1575
Only changes I understood and knew would not break anything else. 
See .. diff: https://github.com/JuliaRobotics/IncrementalInference.jl/compare/22Q3/bug/1574_test..22Q3/bugfix/twig_1574

close #1574
closed 2022-07-26 by dehann