JuliaRobotics/IncrementalInference.jl #1595: test initVariable when using Neo4jDFG driver

dehann, 2022-08-11:


Affie, 2022-08-12:
The short answer is there is nothing wrong with Neo4jDFG. 

@dehann OR @jim-hill-r 

How have you tried to update it?

`initVariable` is only implemented in-memory in IIF so it should be added to DFG or you can do something like this:
```julia
v = getVariable(dfg, :a)
initVariable!(v, Normal(1.0,1.0))
updateVariable!(dfg, v)
```

dehann, 2022-08-12:
Oh, of course. Curse words.  Thanks Johan

dehann, 2022-08-12:
That's why the symmetric design philosophy is important (same behavior regardless of driver is the desire).  We just lost 6 weeks on this front for a potential big customer: 
- https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/issues/264

jim-hill-r, 2022-08-12:
My bad as well for not having a robust test for the numerics which delayed us a few weeks. That is definitely a point of improvement for me.

dehann, 2022-08-12:
Just made the PR.

No worries on your end Jim, its a DFG design philosophy that I've been driving for a while and we just ran into a case where the symmetry between drivers are not exact.  Partly why im always pushing for total symmetry (even between DFG and SDK), but we'll eventually get those in alignment

closed 2022-08-12 by dehann