JuliaRobotics/IncrementalInference.jl #1603: tag v0.30.5

Affie, 2022-08-19:
@JuliaRegistrator register branch=release/v0.30

JuliaRegistrator, 2022-08-19:
Error while trying to register: "Tag with name `v0.30.4` already exists and points to a different commit"

Affie, 2022-08-19:
@JuliaRegistrator register branch=release/v0.30

JuliaRegistrator, 2022-08-19:
Registration pull request created: [JuliaRegistries/General/66561](https://github.com/JuliaRegistries/General/pull/66561)

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if the [Julia TagBot GitHub Action](https://github.com/marketplace/actions/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v0.30.5 -m "<description of version>" ca6f75b6f221d3871ce74182f676f5dc782185ba
git push origin v0.30.5
```

closed 2022-08-19 by Affie