JuliaRobotics/IncrementalInference.jl #1624: initAll! has issue with multihypo recipe in CCW

dehann, 2022-09-05:
New issue on `initAll!`

Likely related to how multihypo recipes are transmitted between CCW and CF.  Might be an old bug uncovered during the recent cleanup of FMd and CPT

```julia
[ Info: init with useinitfct [:x0x2f_9512]
[ Info: do init of x2
ERROR: AssertionError: expected hyporecipe.certainidx to be the same as cached in ccw
Stacktrace:
  [1] computeAcrossHypothesis!(ccwl::WARNING: both ApproxManifoldProducts and TransformUtils export "rotate!"; uses of it in module RoME must be qualified
CommonConvWrapper{Pose3Pose3{FullNormal}, Tuple{Vector{ArrayPartition{Float64, Tuple{Vector{Float64}, Matrix{Float64}}}}, Manifolds.SemidirectProductOperation{Manifolds.RotationAction{TranslationGroup{Tuple{3}, ℝ}, 

    @ IncrementalInference ~/.julia/dev/IncrementalInference/src/services/EvalFactor.jl:380
  [3] #evalPotentialSpecific#349
    @ ~/.julia/dev/IncrementalInference/src/services/EvalFactor.jl:560 [inlined]
  [4] evalFactor(dfg::GraphsDFG{SolverParams, DFGVariable, DFGFactor}, fct::DFGFactor{CommonConvWrapper{Pose3Pose3{FullNormal}, Tuple{Vector{ArrayPartition{Float64, Tuple{Vector{Float64}, nullSurplus::Float64, dbg::Bool, skipSolve::Bool, _slack::Nothing)
    @ IncrementalInference ~/.julia/dev/IncrementalInference/src/services/EvalFactor.jl:589
  [5] approxConvBelief(dfg::GraphsDFG{SolverParams, DFGVariable, DFGFactor}, fc::DFGFactor{CommonConvWrapper{Pose3Pose3{FullNormal}, Tuple{Vector{ArrayPartition{Float64, Tuple{Vector{Float64}, measurement::Vector{Tuple}; solveKey::Symbol, N::Int64, nullSurplus::Float64, skipSolve::Bool)
    @ IncrementalInference ~/.julia/dev/IncrementalInference/src/services/ApproxConv.jl:18
  [6] #calcProposalBelief#359
    @ ~/.julia/dev/IncrementalInference/src/services/ApproxConv.jl:175 [inlined]
  [7] proposalbeliefs!(dfg::GraphsDFG{SolverParams, DFGVariable, DFGFactor}, destlbl::Symbol, factors::Vector{DFGFactor{CommonConvWrapper{Pose3Pose3{FullNormal}, Tuple{Vector{ArrayPartition{Float64, dens::Vector{ManifoldKernelDensity}, measurement::Vector{Tuple}; solveKey::Symbol, N::Int64, nullSurplusAdd::Float64, dbg::Bool)
    @ IncrementalInference ~/.julia/dev/IncrementalInference/src/services/ApproxConv.jl:261
  [8] propagateBelief(dfg::GraphsDFG{SolverParams, DFGVariable, DFGFactor}, destvar::DFGVariable{Pose3}, factors::Vector{DFGFactor{CommonConvWrapper{Pose3Pose3{FullNormal}, Tuple{Vector{ArrayPartition{Float64, 
    @ IncrementalInference ~/.julia/dev/IncrementalInference/src/services/GraphProductOperations.jl:31
  [9] doautoinit!(dfg::GraphsDFG{SolverParams, DFGVariable, DFGFactor}, xi::DFGVariable{Pose3}; solveKey::Symbol, singles::Bool, N::Int64, logger::Logging.ConsoleLogger)
    @ IncrementalInference ~/.julia/dev/IncrementalInference/src/services/GraphInit.jl:129
 [10] #doautoinit!#29
    @ ~/.julia/dev/IncrementalInference/src/services/GraphInit.jl:174 [inlined]
 [11] initAll!(dfg::GraphsDFG{SolverParams, DFGVariable, DFGFactor}, solveKey::Symbol; _parametricInit::Bool, solvable::Int64, N::Int64)
    @ IncrementalInference ~/.julia/dev/IncrementalInference/src/services/GraphInit.jl:502
 [12] initAll! (repeats 2 times)
    @ ~/.julia/dev/IncrementalInference/src/services/GraphInit.jl:449 [inlined]
 [13] top-level scope
    @ REPL[10]:1
```

dehann, 2022-09-05:
Hi @jim-hill-r , is this perhaps the same issues you were seeing when solving on IIF master?

dehann, 2022-09-05:
Got reason to believe this error is not showing up on latest stable release v0.30, so is specific to master branch

dehann, 2022-09-06:
if so, then this might have something to do with test failures in RoME and:
- #1570

Affie, 2022-09-06:
Should I take this one? Do you have an MWE?

dehann, 2022-09-06:
HI Johan, yes please.  Will send you a data file as MWE

Affie, 2022-09-06:
Unfortunately, the MWE you send me works.

Affie, 2022-09-06:
Without having an example to work with and knowing what is done, I can only guess it has to do with unpacking of factors.
My guess would be something to do with `reconstFactorData`

dehann, 2022-09-06:
ah, so the error occurred using NavAbilityCaesarExtensions to make a local copy of the graph.  The FileDFG version does work.

If its coming from a lack of or improper use of `reconstFactorData` then this is likely the source:
https://github.com/NavAbility/NavAbilityCaesarExtensions.jl/blob/a65854b945978993370b738dc5e92c06bed606fb/src/services/Factor.jl#L7

Will send a better MWE example...

dehann, 2022-09-06:
Going to close here and open at the other repo

closed 2022-09-06 by dehann