JuliaRobotics/IncrementalInference.jl #1760: fix nonparametric heading test issue (in RoME.jl)

dehann, 2023-07-21:
See `@test_broken` here:
-

dehann, 2023-10-20:
try to do for v0.35.0, would like to ensure partials working properly for RoME.  Small chance this punts to v0.35.1 given other priorities needing v0.35.0.

dehann, 2023-10-20:
spike and bump to v0.35.1 if taking too long

Affie, 2023-10-25:
Testing: So far `approxConvBelief` gives the correct answers.

dehann, 2023-10-25:
this is new test coverage, and test looks at implicit partial priors and graph is still fully defined.  Decision on how to handle these partials either explicit or code improvements with Manifolds.

Way back before Manifolds and StaticArrays this would work because the implicit partial was handled automatically since all compute and variable assignments were done in coordinates and using manifold dims of variables and factors (i.e. xDim and zDim).

dehann, 2023-10-25:
requires new design session on partials, which expected to occur in the near future

dehann, 2024-11-27:
There has been some progress in understanding new design for partials, using dispatch with named dimensions during defVariable.  Work still ongoing so going to delay in favor of sooner compat resolution in v0.35.5