JuliaRobotics/IncrementalInference.jl #1773: tag v0.34.1

dehann, 2023-09-05:
@JuliaRegistrator register branch=release/v0.34

JuliaRegistrator, 2023-09-05:
Registration pull request created: [JuliaRegistries/General/90839](https://github.com/JuliaRegistries/General/pull/90839)

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if the [Julia TagBot GitHub Action](https://github.com/marketplace/actions/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v0.34.1 -m "<description of version>" 3a3f9e3b739ec2f7bc9fb2ed75e0c46853ee2fbd
git push origin v0.34.1
```

dehann, 2023-09-05:
@JuliaRegistrator register branch=release/v0.34

JuliaRegistrator, 2023-09-05:
Registration pull request updated: [JuliaRegistries/General/90839](https://github.com/JuliaRegistries/General/pull/90839)

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if the [Julia TagBot GitHub Action](https://github.com/marketplace/actions/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v0.34.1 -m "<description of version>" 3a3f9e3b739ec2f7bc9fb2ed75e0c46853ee2fbd
git push origin v0.34.1
```

closed 2023-09-05 by dehann