JuliaRobotics/IncrementalInference.jl #1781: dev issue RoME tests with manopt

dehann, 2023-10-11:
since mergining:
- #1778 

On `using RoME`

```
ERROR: [ Info: monitorCSMs: all tasks done
LoadError: type CalcFactorResidual has no field manifold
Stacktrace:
  [1] getproperty
    @ Base ./Base.jl:37 [inlined]
  [2] (::IncrementalInference.CalcFactorResidual{RoME.PriorPose2{Distributions.FullNormal}, Nothing, 3, 9, SubArray{Float64, 1, RecursiveArrayTools.ArrayPartition{Float64, Tuple{Vector{RecursiveArrayTools.ArrayPartition{Float64, Tuple{StaticArraysCore.SVector{2, Float64}, StaticArraysCore.SMatrix{2, 2, Float64, 4}}}}, Vector{StaticArraysCore.SVector{2, Float64}}}}, Tuple{Vector{Int64}}, false}, RecursiveArrayTools.ArrayPartition{Float64, Tuple{StaticArraysCore.SVector{2, Float64}, StaticArraysCore.SMatrix{2, 2, Float64, 4}}}, 1})(m::RecursiveArrayTools.ArrayPartition{Float64, Tuple{StaticArraysCore.SVector{2, Float64}, StaticArraysCore.SMatrix{2, 2, Float64, 4}}}, p::RecursiveArrayTools.ArrayPartition{Float64, Tuple{StaticArraysCore.SVector{2, Float64}, StaticArraysCore.SMatrix{2, 2, Float64, 4}}})
    @ RoME ~/.julia/dev/RoME/src/factors/PriorPose2.jl:41
  [3] (::IncrementalInference.CalcFactorResidual{RoME.PriorPose2{Distributions.FullNormal}, Nothing, 3, 9, SubArray{Float64, 1, RecursiveArrayTools.ArrayPartition{Float64, Tuple{Vector{RecursiveArrayTools.ArrayPartition{Float64, Tuple{StaticArraysCore.SVector{2, Float64}, StaticArraysCore.SMatrix{2, 2, Float64, 4}}}}, Vector{StaticArraysCore.SVector{2, Float64}}}}, Tuple{Vector{Int64}}, false}, RecursiveArrayTools.ArrayPartition{Float64, Tuple{StaticArraysCore.SVector{2, Float64}, StaticArraysCore.SMatrix{2, 2, Float64, 4}}}, 1})(p::RecursiveArrayTools.ArrayPartition{Float64, Tuple{Vector{RecursiveArrayTools.ArrayPartition{Float64, Tuple{StaticArraysCore.SVector{2, Float64}, StaticArraysCore.SMatrix{2, 2, Float64, 4}}}}, Vector{StaticArraysCore.SVector{2, Float64}}}})
    @ IncrementalInference ~/.julia/dev/IncrementalInference/src/parametric/services/ParametricManopt.jl:76
  [4] (::IncrementalInference.var"#546#547"{RecursiveArrayTools.ArrayPartition{Float64, Tuple{Vector{RecursiveArrayTools.ArrayPartition{Float64, Tuple{StaticArraysCore.SVector{2, Float64}, StaticArraysCore.SMatrix{2, 2, Float64, 4}}}}, Vector{StaticArraysCore.SVector{2, Float64}}}}})(f::IncrementalInference.CalcFactorResidual{RoME.PriorPose2{Distributions.FullNormal}, Nothing, 3, 9, SubArray{Float64, 1, RecursiveArrayTools.ArrayPartition{Float64, Tuple{Vector{RecursiveArrayTools.ArrayPartition{Float64, Tuple{StaticArraysCore.SVector{2, Float64}, StaticArraysCore.SMatrix{2, 2, Float64, 4}}}}, Vector{StaticArraysCore.SVector{2, Float64}}}}, Tuple{Vector{Int64}}, false}, RecursiveArrayTools.ArrayPartition{Float64, Tuple{StaticArraysCore.SVector{2, Float64}, StaticArraysCore.SMatrix{2, 2, Float64, 4}}}, 1})
    @ IncrementalInference ~/.julia/dev/IncrementalInference/src/parametric/services/ParametricManopt.jl:172
```

Affie, 2023-10-12:
Maybe this should be a design discussion.
I would like factors to call `getManifold` or hard code the Manifold for now.  It used to be like this in most of the factors and I've reverted it where I saw. The reason is to first simplify CalcFactor as much as possible to get rid of dynamic dispatch which is a major bottleneck in performance and then add fields as needed. If done correctly, getManifolds uses constant propagation during compile time and is faster than getting a field in a struct.

Affie, 2023-10-12:
Regardless, it looks like getManifolds is faster: 
```
julia> @btime getManifold(cf)
  13.105 ns (0 allocations: 0 bytes)
SpecialEuclidean(2)

julia> @btime cf.manifold
  77.283 ns (1 allocation: 544 bytes)
SpecialEuclidean(2)
```

closed 2023-10-20 by Affie